procount/pinn #113: Automatic HDMI / DSI detection not working with USB booting

solarsootysmudge, 2018-01-06:
**Problem** : Automatic HDMI / DSI detection not working with USB booting. When changes are made in either config.hdmi or config.dsi. The changes do not happen in the Config.txt at runtime. This only seems to be a problem when using a USB. Whilst SD cards work fine.

**Steps to replicate** : using a Pi that is can boot from USB or SD card (See https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/bootflow.md )

Install a brand new copy of PINN on a FAT32 formatted USB stick and SD card. Install Raspbian on both and then go through steps in this forum post https://www.raspberrypi.org/forums/viewtopic.php?f=108&t=150889

The SD card will trigger auto detection of the HDMI cable or DSI screen and write changes to OS config.txt. Power down remove SD card and plug in USB stick. The USB stick will not detect changes in screens.

Given the nature of the problem this will require more testing on my part. Help required in checking step to replicate please.

procount, 2018-01-06:
I'll take a look.

procount, 2018-01-06:
Ahh, I see the problem - it was not adapted for USB. I'll see if I can fix it in the next version. Thanks for spotting this.

procount, 2018-01-13:
fixed in p2.4.5

closed 2018-01-13 by procount