procount/pinn #114: Add additional recovery keys to as well as shift

procount, 2018-01-08:
https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=142574&p=1255283#p1255284

Re: STICKY: PINN - An enhanced version of NOOBS.
Report this post Quote
Post  by efa » Sun Jan 07, 2018 12:04 am

PINN has the ability to change the recovery key instead of SHIFT via text file option?
I'm using this controller for media player:
https://www.amazon.it/gp/product/B008RTE16K
that has numbers, letters, CAPS/NUM LOCK, cursor key, mouse, ENTER, ESC, and some other keys, but miss SHIFT.
I read about using jumper or forcetrigger option at:
https://github.com/raspberrypi/noobs/bl ... oard-input
but certainly changing the recovery trigger key is better.

procount, 2018-01-17:
Suggested on thread to use mouse function to left-click on the raspberry logo

MartinX3, 2018-02-08:
That would be nice.
I use remmina to use the vnc function.
I am not able to send SHIFT.

I'm also not able to use STRG+ALT+F1 etc, because it is captured from my linux system.