procount/pinn #115: Install the same OS multiple times from online sources

EctoOne, 2018-01-08:
I want to install multiple instances of the same OS on one card. For example, one Raspbian for Job related stuff and one for Private Stuff.

I've took a quick look at noobsconfig but it seems to me that it's only for installations with SD/USB images. So far I managed it to install the same OS twice, by downloading an image and alter the files to have it show up as separate system.
I would prefer if I could install (reinstall) them from online sources. So I'm wondering if it's possible to only put a minimum of files (files to determine the name, description, boot and root labels. And maybe an different icon) on the SD.

If that is somehow possible it would be great if I could get an example on how to do it.

Now that I think of it, it would be nice to have that option within PINN. Kinda like most windows installer work. They ask for a path and such.

procount, 2018-01-08:
You need to use a flavours.json file to describe the different instances of the same OS, but in reality there doesn't need to be any difference between the OS versions. With the latest version of PINN this should allow you to reinstall any one of them without affecting the others. Have a look at Retropie for an example of what you need to provide.

You'll probably need to download the OS, add the flavours.json and png files, and then install from USB to get it to work properly. I'm looking to improve this (and the noobsconfig feature that was built on top of this) to allow remote installation and to separate the customisations from the OS files. Boot and Root labels will be automatically altered for each instance.

EctoOne, 2018-01-09:
So as of now, I won't work if I just place an flavors.json with the added custom variant and the png inside a folder with the system I want to duplicate. And then put that in PINNs OS folder on the SD card, am I right? I have to provide all files via USB or SD card.

procount, 2018-01-09:
Putting it in PINN's  os folder will work too.

EctoOne, 2018-01-09:
OK, i did some testing on a small SD card i had.
After some issues with the files I downloaded from sourceforge, I downloaded Retropie directly using PINN. I then modified the flavors.json from the sourceforge folder and replaced the DSI flavor in it (since the DSI files weren't downloaded by PINN anyways), and i also added a renamed/modified png.

I keep the folder on the USB Stick and both systems showed up. Great.
I then moved the Retropie2 folder to the OS folder on the SD and they showed up again. Even Better.

The i deleted both tar files to see if it would work, if the files were downloaded because in some file they are linked. But that gave me an error about missing files. Too Bad.

Then I tried installing them with all files available, and it worked. Even reinstalling worked fine. So yeah, i guess i have to keep the files on my SD for now.
But maybe that can be a good thing, because i can update/prepare my system and then create new tar packages which i can then use if I ever want to reinstall any one of those systems. Saves me at least the trouble to completely prepare a system when i had to reinstall it from an online source.

closed 2018-01-12 by EctoOne

procount, 2018-01-12:
If you want the os tar files on the web, but the custom flavour on your sd card, you can do it but you need to add "tarball" options to partitions.json.

I'm revising this flavour customisation at the moment so keep an eye out for updates.

EctoOne, 2018-01-12:
Awesome support! Although I'm fine now. I've created my own tarballs from a pre-configured system and moved them on my SD card. Since it's a 32gb card I don't mind losing the space. Maybe I will use it someday. For now I can't wait for the latest update to be available via auto update, because changing the background color was something I would have asked for too.