procount/pinn #117: Uninstall OS

DyingEcho, 2018-01-24:
How can I uninstall an OS I previously selected to free up space on the SD card?

procount, 2018-01-24:
I'm afraid that is not supported.
You have 2 options: 1) delete everything by reinstalling only the OSes you need. 2) manually remove and resize partitions using a tool such as gparted and adjust /settings/installed_os.json accordingly. The 2nd option is an advanced operation though.

DyingEcho, 2018-01-25:
@procount How might I do this? I have Raspbian and Arch. Would I erase the `boot` partition, leaving `boot_arch` and `RECOVERY`? Or would I need to do something with `disk2s7` or `disk2s9`, shown below? 
[image]

procount, 2018-01-25:
I'm sorry but I have no experience on Apple products so I don't know how to help on that platform.
This is quite an advanced topic, so I would seriously recommend backing up whatever partitions you want to keep to a USB stick (using tar, cp or rsync), re-install whichever OS you want afresh, then restore your backed up partitions over the top of it. (You may want to keep the/etc/fstab from the new installation and restore that after restoring your backup to ensure your OS works with the new partition numbering).

If you _really_ want to delete the unwanted partitions, then you  will have a lot of "googling" to do to understand how to mess about with partitions and learn the tools to do it on a Mac.  And also understand how to edit /settings/installed_os.json (not so difficult). You should have a look at https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained, especially noting the partition layout _after_ v1.5.

closed 2018-04-11 by procount