procount/pinn #119: Ubuntu Server is not Booting

rellik8821, 2018-01-26:
Hello,
i add ubuntu 16.04 server i use this image:
ubuntu-16.04-preinstalled-server-armhf+raspi3.img

But is not booting my screen looks like:
https://i.stack.imgur.com/zUZn6.jpg

any idea how to fix?

procount, 2018-01-26:
Is this a 64-bit image?
It uses Uboot as a boot loader so you need to tell uboot where the rootfs is.
Modify boot.scr

rellik8821, 2018-01-26:
is raspi3 preinstalled-server image.

dont have a boor.scr
try create rootfs with name cloudimg-rootfs

procount, 2018-01-26:
I don't know either.
I haven't got a uboot system working yet. Each one seems to be different.

rellik8821, 2018-01-26:
ok.
what is  the problem?

looks like the system want boot from network because can' finde rootfs?

procount, 2018-01-27:
Correct.
Normally you have to modify boot.scr.
It is probably _inside_ the uboot image.

rellik8821, 2018-01-27:
I do not know what to modify in the boot.scr.
if anyone is interested here has the OS

https://www.dropbox.com/s/l2d6dbe080ufmjf/UbuntuServer.zip?dl=0

procount, 2018-01-27:
It looks like you may be able to side-step UBoot if you look at:
https://wiki.ubuntu.com/ARM/RaspberryPi  & search for  "Booting the official Pi 2 image on the Pi 3"

rellik8821, 2018-01-27:
i have set some changes image will boot without PINN

with PINN i',m in emergency mode.

First:
https://imgur.com/CkE92Xv

after /1min 30s
https://imgur.com/1SqvJyL

closed 2025-10-17 by procount