procount/pinn #120: CEC not working with 2.4.5a with a panasonic TV and remote

tlfwwm, 2018-01-28:
Hi'

I have for several years had the OSMC media center running stand alone on a raspberry pi with the remote control of the TV running flawlessly, so CEC has in other environments been demonstrated to work. Recently I decided to set up a dual boot raspberry pi with PINN as the mechanism for selecting, which operating system to boot. The desire is to have this running headless and to control what to boot via the TV remote control

However, when in PINN mode the arrows and OK buttons of the TV remote does not respond and move the cursor on the screen. This goes in general for the buttons on the TV remote. I have logged into recovery mode and only see this of relevance in /tmp/debug.

2.4.5a 
Board revision is  10494082 
CecListener thread started 
CecListener done initializing 
Found recovery.rfs at "mmcblk0p1" 
PINN drive: "/dev/mmcblk0" 

Nothing of immediate relevance to CEC is found in neither /tmp/debug nor in dmesg output and the process table does not show any processes with CEC in it.

Any help in further debugging this matter will be highly appreciated.

Best regards.

Thomas.

procount, 2018-01-28:
There is some explanation here -> https://github.com/raspberrypi/noobs/issues/128

Adding support for programmable CEC keys is on my development todo list.

tlfwwm, 2018-01-28:
Hi'

Just for clarification, does this mean that the suspicion is that programmable CEC keys will eventually be the answer to my prayers :-) :-)? Or, is there some other measures or tricks, which I can try to apply meanwhile?

Best regards.

Thomas.

procount, 2018-01-28:
Hopefully.
It seems CEC is a standard that every TV manufacturer implements differently and calls it by another name.
I know my codes work with Samsung but I guess Panasonic is different.
So I need a way to expose the key codes your tv uses so I/you can map them to PINN controls.
It seems a lot like LIRC where every remote sends out different codes.
Incidentally, when you get the raspberry logo come up on PINN where you normally press shift to enter the menus, does pressing any key on your remote do the same thing?
And does your TV recognise your RPi as PINN alongside the hdmi device name?

tlfwwm, 2018-01-28:
Hi'

No response to key presses on the remote in any views so far. I have a mouse attached to get into the recovery menu, but there I also have no response to key presses on the remote.

The TV is somewhat dated, which is presumably also why it only shows HDMI<nr> in the AV input selection menu. I have another HDMI<nr> port attached to an OSMC box and that does not show as OSMC either. Still, though, the remote works fine there.

Best regards.

Thomas.

procount, 2018-01-28:
Pinn v2.4.5d has just been released. There is a small change in there that might help you.
Once installed, and you've got to the recovery menu, press ctrl-alt-f2 and login with root/raspberry.
Then type `tail -f /tmp/debug` which will display some debug messages.
Try pressing the keys on your TV remote and see if you get any lines beginning `CEC Key:` and note down the numbers following, along with the key you pressed on the remote.

tlfwwm, 2018-01-29:
Hi'

Hmmm, fetched /archive/pinn-245d.zip from sourceforge and extracted it over the recovery partition, taking good care to have a backup copy of recovery.cmdline and putting that back in place afterwards. Then booted and entered the recovery menu by left clicking on a still attached mouse. The menu bar on the resulting window displays PINN v2.4.5d, so indeed things are as expected.

However, everything to no avail. The head of /tmp/debug is given below, but key presses on the TV remote subsequent to booting into the recovery menu produces no further output to /tmp/debug.

# cat /tmp/debug 
2.4.5d 
Board revision is  10494082 
CecListener thread started 
CecListener done initializing 
Found recovery.rfs at "mmcblk0p1" 
PINN drive: "/dev/mmcblk0" 
Tap detected 
json_parser - syntax error found,  forcing abort, Line 1 Column 1 
Error parsing json file: "/mnt/overrides.json" 
Starting dbus 
Starting dhcpcd 

No interesting output from dmesg or messages is found as well. This is strange since indeed does a raspberry pi with OSMC work all fine with the TV remote.

Best regards.

Thomas.

procount, 2018-01-29:
You could have just let it self-update ;)

Oh well it was worth a try. I have one more optiom to try... but in any case your debug log indicates I may have made a tiny error elsewhere, so I'll fix that. Thanks.

procount, 2018-01-29:
I put a copy of pinn-245dt.zip in the /testing folder. Please try that one, taking the same care you did last time. See if the /tmp/debug listing shows any "Vendor key:" entries.

tlfwwm, 2018-01-29:
Hi'

Bummer, tried both 245dt and also adjusted config.txt, both things to no avail. Taking note that the recovery window still displays 245d and not 245dt but have cross checked that it is indeed the file of the latter zip file, which I put on the sd card. Hmmm, troublesome, but thanks for trying to help nail it down.

Best regards.

Thomas.

tlfwwm, 2018-01-30:
Hi'

Interesting evolution. I installed OSMC via PINN and I was fully successful in configuring OSMC via my TV remote. Now, after a warm reboot of my pi from OSMC, key presses are successfully detected by PINN during initial boot. Thus, it works, I can move the cursor around on the screen, and I have detected the following cec key responses.

CEC key:  1 : Up
CEC key:  2 : Down
CEC key:  3 : Left
CEC key:  4 : Right
CEC key:  0 : Enter

Number keys and channel up and down are not detected. The exit key is detected with cec code 44. The return key is detected with cec code 13 and appears to reboot the pi.

Subsequent booting into retropie from PINN followed by a warm reboot still makes PINN detect key presses from my TV remote.

However, if I turn off the pi and TV completely and then turn on again, then my TV remote key presses are not detected. However, if I boot into OSMC, then my TV remote works and, again, with a warm reboot the key presses on the TV remote is detected by PINN.

So, it would seem that CEC keys are mapped fine but that initialization is not working from a cold boot. Interestingly can OSMC get to activate the CEC communication, which then stays functioning, provided that subsequent boots are warm.

Unfortunately, this all gets lost after having the TV in standby for a short while, so this would only be sustainable if the TV would always be turned on. Thus it seems that the culprit is really how to get a proper initialization of the cec communication.

Best regards.

Thomas.

procount, 2018-01-30:
Yes, I think you are right. As I said, CEC is not standardized between vendors, but LibCEC (which all the Kodi distros seem to use) does a good job of making them all appear the same. I haven't used it for the same reason that NOOBS/Berryboot don't use it. So maybe I'll have to change that decision or do something similar. But it's not easy when I don't have a Panasonic TV to test it.
In OSMC etc, do the number keys and channel keys work there? I read somewhere about having to send some additional codes to the TV to enable that feature.

procount, 2018-05-11:
PINN v2.8.1 now has programmable CEC buttons

closed 2018-05-11 by procount