procount/pinn #121: Access through ssh/vnc while at boot selection menu.

Landberg, 2018-01-30:
I was hoping I could use PINN as a remote recovery tool, but I can't seem to get access to ssh and/or vnc at the boot selection menu unless I've pressed SHIFT during boot.  **vncshare** and **ssh** are activated.

As I use the **bootmenutimeout=60**, I expected to get the same behavior as when I manually pressed SHIFT, to enter the recovery GUI, then pressed ESC (exit) to enter the boot menu. My guess is that you don't launch any ssh/vnc-services unless I have actively selected the recovery GUI.

I still want it to auto-select the OS and boot to it as default, but would like to have a little time (bootmenutimeout for example) to force-boot to another OS or the recovery-GUI.

I have a headless remote box that on some occasions fail to boot properly. It can be a failed upgrade or botched config edit for example. As it is remote, I can't press any keys during boot.

If I do it manually (press SHIFT then ESC) I get the perfect option to ssh into it and do a `su -c 'echo 8 > /sys/module/bcm2709/parameters/reboot_part';reboot` to reboot into my second OS and from there mount the first OS and fix the config error for example.

Have I missed some option to always have ssh/vnc active during boot menu?

If it's not possible, it could perhaps be implemented by adding an optional timeout to the GUI that in combination with **forcetrigger** would continue with **ESC** to exit after the specified GUI-timeout unless the mouse is moved, key is pressed or something? I don't mind if it takes a minute or two to get to the boot menu, as long as I have access to ssh once I'm there. 

Keep up the good work!

procount, 2018-01-30:
Ah, I see what you mean. The ssh/vnc services are launched, but I don't think the network is started unless you go to the recovery menu with forcetrigger.
I'll see what I can do

procount, 2018-01-31:
Please download pinn-245e.zip from /testing.
You'll need a password to unzip it. Can you guess what it is?
recovery.cmdline should have `ssh vncshare forcetrigger remotetimeout=20` in it.  (Adjust remotetimeout value to suit an appropriate delay.)

So, forcetrigger goes into the recovery menu to start the network.
remotetimeout sets a countdown timer in seconds, after which it will go into the boot menu and start whatever countdown timer is set there. The countdown timer is shown in the window title.
Any key or mouse press will stop the countdown timer, allowing you to change the OS to boot.

Landberg, 2018-02-01:
Fantastic!
I've done a couple of tests and it works great!
Thank you very much!

procount, 2018-02-01:
Well, not quite perfect yet, so there may be some little changes for the next version, but I think it is mostly there.

closed 2018-02-01 by procount