procount/pinn #127: VNC > Update PINN Dialogue in Recovery doesn't stop the timer

MartinX3, 2018-02-08:
The Update PINN Dialogue in Recovery doesn't stop the timer, when I am connected via vnc and move a window or click on it.
It will open the choose OS list.
I don't want to raise the waiting time of 10 seconds, because that would be too long in a regular usage scenario.

procount, 2018-02-08:
Are you using the remotetimeout option?

Check update is in the docs - see https://github.com/procount/pinn/blob/master/README_PINN.md#self-update

MartinX3, 2018-02-08:
@procount 
Yes, I use remotetimeout to get 10 seconds in the recovery screen, forcetrigger to open the recoveryscreen and boottimeout to get 10 seconds to choose the OS.
My problem is, that this update-dialogue cause a bug in the "remotetimeout" function.
It should stop the timeout, so i have enough time, to read the changelog and click on "yes/no/etc...".

procount, 2018-02-08:
It's not a bug. The default behaviour is to boot the last OS. If you want it to stop you have to interact with it manually. That is what the remotetimeout is for as per the original request #121

MartinX3, 2018-02-08:
When I connect with VNC into the recovery dialogue and move the mouse or click, the timer get stopped.
That is the normal behaviour.
When this update dialogue appears, the timer doesn't get stopped.
The update dialogue breaks the standard behaviour.
So it is a bug.

procount, 2018-02-08:
Maybe I misunderstood you. I thought you meant the appearance of the update dialog should stop the timeout. Do you mean it prevents you clicking on the main window to stop the timeout? Then you have to close the dialog first before clicking on the main window. There are several dialog boxes that may pop here, not just the update dialog.

MartinX3, 2018-02-08:
Thanks for your help. :)
When I close this dialogue, the timer stops, but how to get this dialogue back, to update PINN?
I want to update PINN, but I also want to stop the timer with my user interaction at the same time, to get enough time, to read the changelog. :)

procount, 2018-02-08:
Did you read the link I gave you above?
Go to Maintenance Menu, select only PINN and click Re-install.

MartinX3, 2018-02-08:
Yes, I read it.
Thank you for sharing this information with me. :)
This way isn't really intuitive, but it works.
(I didn't thought, that a "reinstall" means "upgrade" especially after getting a warning dialogue about data loss).

But the issue I'm reporting is, that the behavior of the timer is broken when getting this upgrade dialogue.
The timer which opens the choose os menu should stop, when I move the upgrade dialogue or interact on another way with this screen and a mouse.

It works fine, when this dialogue doesn't appear. I move my mouse and the timer stops, so I get the time i want, when I manually interact with the GUI.
So it should also behave on the same way, when this upgrade dialogue comes.

procount, 2018-05-15:
Fixed in next release.

closed 2018-05-15 by procount

MartinX3, 2018-05-15:
Thank you, but shouldn't I test it before you close this issue?

procount, 2018-05-15:
Already tested, but I'm sure you'll let me know if it doesn't work and then I can reopen this issue.