procount/pinn #128: RasPi keeps freezing when loading PINN

Pyroseza, 2018-02-09:
I have downloaded the latest PINN and am unable to get any OS downloaded and running.

**My Raspberry Pi is a model 3 B, it keeps freezing on boot.**

I have tried a few different methods combinations of power supplies (including the official power supply, 2 battery packs, iPhone charger and Android Tablet charger). I have tried different display methods (HDMI, Official 7 inch touch screen with ribbon adapter and the Pimoroni Hyperpixel connected via GPIO pins). I have gone through 4 SD cards now all of which now have PINN flashed on to (2 of which I got from Raspberry Pi with NOOBS preloaded, 1 Samsung EVO 16GB I used to use on my phone and now a brand new San Disk Ultra 16 GB). All of these cards are class 10, there is nothing wrong with them.

Does anyone know why the Raspberry Pi keeps freezing like this? It used to work fine before I went and flashed NOOBS?

procount, 2018-02-09:
I have occasionally experienced a freeze whilst installing an OS from the web, but a reboot and retry normally fixes it. It is quite rare for me.
When exactly is it freezing? You said "when loading pinn" in the title and "freezing on boot" in bold - booting what exactly?
If freezing during installing an OS, have you tried downloading an os to a usb stick and installing from there?
I'm interested to hear that you tried it with the hyperpixel. Apart from the freezing, did it work straight away, or with a Pimoroni driver installed?

Pyroseza, 2018-02-13:
Hi, so it freezes in a few places. 1. shortly after pinn has loaded the kernel for the OS, 2. sometimes it loads all the way in to the OS login screen (Kali & Raspbian). I have noticed that it freezes earlier if I have the keyboard’s wireless adapter plugged in. I have a wired usb keyboard as well but that I think draws too much power for the power, so it freezes sooner. Basically it is always freezing after only a few seconds of booting. I did not have this problem when I had noobs installed before, but now if I try that route it freezes no matter what, OS loaded directly, noobs booting the OS or PINN booting the OS. 

For the hyperpixel I had to install the Pimoroni driver. But I did that using another raspi. 

For some reason after installing PINN and putting it in this raspi it has has somehow messed it up and won’t stay running.

Pyroseza, 2018-02-13:
To add, if I hold down shift and load up recovery mode or stay on the PINN screen where you can choose the OS it will freeze within a minute, I tried timing it, it’s not the exact same time every time. I have sufficient cooling on it: thermal paste, alluminium cooling block on the CPU and a 40mm dc fan plugged into a separate usb power supply faced directly on the block. Not sure what it can be.

Pyroseza, 2018-02-13:
Here is an example of it freezing on PINN initialisation
[image]

Pyroseza, 2018-02-13:
In Kali it freezes on the login screen
[image]

procount, 2018-02-13:
When PINN exits and you boot into another OS like Kali, then PINN is completely out of the way. It has just told the Pi to reboot into a different partition, so it's like starting from a power-on condition as if PINN wasn't there. So I don't think it is PINN that is causing the problem. I think you might have another hardware issue going on. Have you tried flashing your SD card with Raspbian using Etcher and trying that to see if that freezes too? Might be a good test, but I suspect it will be the same result, especially as it is happening with NOOBS as well. There is nothing that NOOBS or PINN can do to the RPi that could cause such an effect.

Some of my tests I do on a Pi3, powered by the USB port of my laptop, with the little lightning symbol indicating low voltage. No heat sink, no fan etc. and it still works fine without freezing.

Pyroseza, 2018-02-13:
Yeah I suspect it’s something else. It just seems to be coincidental that it happened after loading PINN. Anyways do you know where I can start looking for the problem? I managed to somehow load into a shell earlier and that was working perfectly. Seems like it’s a GUI related thing...

procount, 2018-02-13:
I have no idea, I'm afraid.
I assume it still freezes without the hyperpixel attached?
You said you used another Raspi to load the pimoroni driver. You could try swapping cards between the 2 raspi to rule out the sdcard & software being the cause.
If it still freezes with plain raspbian flashed with etcher, then you could ask on the forum. Most people are familiar with that combination and it will help people focus on your problem rather than how you booted the os!
Did you attach any hardware (hats) around the time it started freezing?
Other than those points I'm at a loss.

procount, 2018-02-13:
Try booting plain raspbian to the command line (use raspi-config). If that is stable for a while, try typing startx to get to the desktop and see if it then freezes. That will test your gui theory.

Pyroseza, 2018-02-13:
okay just made a break through, grabbed a new sd card, flashed raspbian directly using etcher, like I did before on other card and it started up with a black screen, I pulled the power and booted it up again then all of a sudden it was working. I then rebooted to see if it is fine and I'm getting the black screen again... I am so confused right now

Pyroseza, 2018-02-13:
okay got it working on the 7 inch touch right now. just rebooted a few times till it worked
I really don't know why this is happening but I can't get it to work every time.
I really want PINN working on it...

procount, 2018-02-14:
If it's the 7 inch RPF touchscreen, make sure the ribbon cable connections are good. The screen is only initialised at power on, so if it doesn't work straight away, it will not work until it is rebooted again.
PINN works fine on mine.

Pyroseza, 2018-02-14:
The 7 inch screen is fine at boot does exactly the same as my monitor.
Oh well this could be some underlying hardware issue. I’ll look at getting another pi. Anyways I’ll close this issue off, thanks for replying. Cheers

closed 2018-02-14 by Pyroseza