procount/pinn #129: No Network Available with LAN

MartinX3, 2018-02-24:
I get the "No Network Available" message.
But the Raspi 3B is connected with an Ethernet cable.
LibreElec and Raspbian both have Internet access.

_Sent from my Nexus 6 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

procount, 2018-02-24:
Go to recovery console with
Ctrl-alt-f2 login root password raspberry.
What is the result of ifconfig?

procount, 2018-02-24:
/@MartinX3 - Are you reporting that the "No network available" message is appearing when it shouldn't, because the Pi is connected to the network over Ethernet and PINN has downloaded the list of OSes?
Or are you reporting that the network is not connecting, even though the network cable is physically connected, so you don't get a list of OSes?

MartinX3, 2018-03-04:
Sorry for the delay.

5 Seconds after the error appears, i get the list of the images, because the box is connected with the ethernet cable.

After i closed the error message, it gold me to choose a WLAN network.

The raspberry pi 3 has network.
ifconfig is fine.

Maybe it looks too fast for a connection before giving me this error?

procount, 2018-03-04:
Yes it looks like a timing problem.
Ethernet connections usually connect a lot faster than that. The time is extended if a wifi dongle is detected. Maybe a configurable time delay would be better so you can adapt it for your network speed.

MartinX3, 2018-03-04:
That would be nice.
The next LAN-Switch is a Fritz!Box 7170, which is connected via CAT6 Ethernet to the Router (Fritz!Box 7490).

MartinX3, 2018-03-05:
Will that be a "wait"- or a "timeout"-timer?
I would prefer the Timeout-Timer. :)