procount/pinn #130: Can't install Retropie2

superbob, 2018-02-25:
Using the PINN v2.4.5f, when I try to install to install Retropie2 to my Raspberry Pi 3 SD card I get the following error messages :
Erreur pendant le téléchargement des meta fichiers (error while downloading meta files)
http://download.sourceforge.net/project/pinn/os/Retropie2/Retropie2.png

Erreur pendant le téléchargement des meta fichiers
http://download.sourceforge.net/project/pinn/os/Retropie2/partition_setup.sh

Erreur pendant le téléchargement des meta fichiers
http://download.sourceforge.net/project/pinn/os/Retropie2/marketing.tar

Erreur pendant le téléchargement des meta fichiers
http://download.sourceforge.net/project/pinn/os/Retropie2/partitions.json

Erreur pendant le téléchargement des meta fichiers
http://download.sourceforge.net/project/pinn/os/Retropie2/os.json

Hower I installed successfully Raspbian.

procount, 2018-02-25:
Sourceforge have been updating their data centre this week and I think it has caused some problems. Stars have been down since 14 Feb.
I suspect it will start working again soon.

superbob, 2018-02-25:
Thanks for the information.
Are there simple instructions to install the os manually? (without download)
I've read somewhere it was possible to download files to an usb thumb and install directly from it.

procount, 2018-02-25:
You still have to get it from sourceforge so still may have issues.
There is a download option in Pinn in archive toolbar which is the easiest.
Otherwise you can go to sourceforge pinn page directly using your browser and download the files for whatever os you need to the /os/<osname> folder of your usb stick

procount, 2018-02-26:
Additionally sourceforge experienced a DDOS attack but they report it's all working again now, so maybe give it another try now.

superbob, 2018-02-27:
It has came back to normal.

closed 2018-02-27 by superbob