procount/pinn #131: Allow install to SD card AND USB

superbob, 2018-02-27:
Currently one can install OSes either to SD card OR USB.

It could be nice to have the ability to install some OSes to SD and some other to USB to propose all the OSes in the choice boot menu in normal mode.

This would need to add the choice of "where to install" after choosing the OSes during install mode.

I'm trying to acheive this manually but I haven't succeeded yet.

procount, 2018-02-28:
I think that would be quite tricky, especially having to keep track of the amount of disk space used on 2 devices for partition allocation.

superbob, 2018-03-01:
Maybe this is a too specific need.

I agree, it would be hard to integrate this in the current installer.

As an alternative, I wanted to try this:

Let's say I want to install OS A on SD card and OS B on USB.

1. Install OS B to the USB device using PINN.
2. Keep SD settings partition files (mmcblk0p5) somewhere, especially partitions configuration.
3. Install OS A + OS B to the SD card using PINN. This will rewrite mmcblk0p5. (I suppose they will be install in the order "A then B").
4. Tweak partitions on the SD card with gparted (using another computer):
  i. Delete main partition of OS B (mmcblk0p9)
  ii. Move boot partition of OS B (mmcblk0p8) to the end of the logical volume (mmcblk0p2)
  iii. Resize main partition of OS A (mmcblk0p7) to "eat" all available space left be the move of mmcblk0p8 partition.
5. Change main boot files (mmcblk0p5): make main partition of OS B on USB instead of SD using files kept on step 2.

Do you think this could work?

procount, 2018-03-01:
Manually, you can do what you want. It will work if you know what you are doing. :wink:
When you have moved all the partitions around as you want them, remember to update the installed_os.json file in /settings (/dev/mmcblk0p5) to reflect the partitions where PINN can find the OSes to boot them.
And you may need to update the cmdline.txt and etc/fstab entries so each OS can find their new partition locations. NOTE even if you use partuiids, the partuuid for the USB will be different to the SDcard. The partuuid also includes the partition number.
Don't expect PINN to maintain this setup! The Reinstall option **__might__** work, but test it on something non-critical first.

superbob, 2018-03-25:
I finally managed to have the right setup.
The difficult part was the repartitionning as gparted live wouldn't see my SD card correctly. I'm using a Macbook Pro and I booted it using EFI boot, but once started it complained about the SD card not labelled correctly 😕

I had to use [Eassos Partition Guru Free](http://www.eassos.com/partitionguru-free.php) on Windows which support ext4 partitions and did not complained about my SD card label. I had to manually adjust 4K alignement.

I'm closing the issue. I don't know if you think of adding this a full feature. Maybe it is to specific.

closed 2018-03-25 by superbob