procount/pinn #132: Local images From USB

darklord111, 2018-03-06:
Hello.
Say for example i downloaded an image from official Raspbian website.It is in .img format.Now,What do I do to install it on my RPi 3B with PINN?  I read somewhere that you cannot use these directly.
Please explain in detail, I am a beginner and if possible tell methods that are usable in Winows.
I don't have MacOS.
Help Please.

procount, 2018-03-06:
What OS have you tried to install? Most images from the RPF website are already installable by PINN just by installing PINN and connecting it to the internet.

darklord111, 2018-03-06:
I tried Kali.
The thing is both Raspbian and Kali are 1.6 gigs and I live in India.
It will be very difficult to download both while in PINN.
Hence my need for local images.

darklord111, 2018-03-06:
My internet connection is not very fast,too.
So 3.2 gigs in just Raspbian and Kali plus Kodi and Retropie and Rasplex will be RIP to my connection.

procount, 2018-03-06:
Then you could download the PINN images for kali, Retropie etc manually from [sourceforge](https://sourceforge.net/projects/pinn/files/os/). Use a file downloader that allows resuming after an interruption in case of a download error. Once you have downloaded them, you can put them on a USB memory stick and install them locally. This is documented in the [readme](https://github.com/procount/pinn/blob/master/README_PINN.md#local-images-on-usb-stick)

darklord111, 2018-03-07:
Hey thanks.
BTW which is Raspbian? I can see something called XBian RPi3.
Also There is Kali1 and Kali2.Which should I get?Same for Retropie.

Also the os/ folder on USB stick should be for example Raspbian -
os/Raspbian
And inside Raspbian folder should be the files downloaded from Sourceforge,right?

darklord111, 2018-03-07:
Also,does AIY projects include Picroft?

darklord111, 2018-03-07:
Kali doesn't seem updated.OffSec relased a 2018.1 version.

procount, 2018-03-07:
For the complete list of OSes, follow the links from this file:
https://github.com/procount/pinn-os/blob/master/os/repo_list.json
Read each file for info on which folder to download. Typically a '1' is for pi0 or pi1. 2 is for pi2. 3 is for pu3.
Create a folder called os in root of your usb stick. Copy the complete raspbian folder into os so you get /os^raspbian/(files).
AIYprijects wss the first release of software for the google voice hat. No idea what picroft is.
Sorry if some are a bit out of date. I update them now and again when I have time.

darklord111, 2018-03-07:
Ok thanks. Picroft is the pi version of Mycroft.Mycroft is basically Microsoft's Alexa. Also it's Open Source.
OK thanks again

closed 2018-03-07 by darklord111