procount/pinn #135: Restart network after failed connection

sarossell, 2018-03-15:
PINN rocks! Please excuse my Noobity... Long time listener, first time caller...

I dutifully searched the forum but did not find a solution. My Internet connection is shared from my iPhone through my Mac and it often has breaks in connection. If this happens while the recovery screen is initializing, there is a network disconnect icon displayed by each operating system and the WiFi button is disabled. I connect my Pi by direct LAN, not wifi. There seems to be no way to reinitialize the program or network discovery feature. I would try to use the command line to restart the /usr/bin/recovery program by typing Ctrl+Alt+F2, but I have yet to figure out how to do that on a Mac over VNC (sorry to add an issue within an issue). I did find that if I selected an OS to install in spite of the apparent lack of Internet connectivity, it went ahead and tried and eventually succeeded to download, which leads me to think that it performs another discovery before download. I don't know.

So, my question(s):
1) Is there an easy way to restart network discovery or enable wifi if the button is disabled?
2) BONUS: How do you type Ctrl+Alt+F2 on a Mac in VNC?

Thanks!

P.S.: Useful Tip - VNC Viewer seems to require at least one of the top three "All available colors" options with the Adapt to Network Speed option un-selected in order to view the recovery program. Otherwise, you get a black screen as if trying to connect to a vncserver on a console machine with no GUI installed. (That ate 8 hours of my life to figure out!)

procount, 2018-03-16:
Welcome, and thanks!

> My Internet connection is shared from my iPhone through my Mac and it often has breaks in connection. If this happens while the recovery screen is initializing, there is a network disconnect icon [...]

So, your connection is: Iphone ---> wifi ---> mac ---> ethernet ---> RPI ?

The Wifi button is enabled ONLY if you have a compatible wifi adapter attached to the RPi. Pi0W and RPI3 will have it enabled. Others will have it enabled if you have a wifi dongle attached that PINN supports. It is not dependent on ethernet connection or wifi connection to an access point.

> There seems to be no way to reinitialize the program or network discovery feature.

No you're right. Although, if you disconnect and re-plug the Ethernet cable, it should start a new DHCP discover and that may get your connection going again. It may scan for new OSes, depending on where it got to.

> I would try to use the command line to restart the /usr/bin/recovery program by typing Ctrl+Alt+F2, but I have yet to figure out how to do that on a Mac over VNC (sorry to add an issue within an issue).

You can't get to the recovery screen over VNC. Use SSH instead to access it remotely.

> I did find that if I selected an OS to install in spite of the apparent lack of Internet connectivity, it went ahead and tried and eventually succeeded to download, which leads me to think that it performs another discovery before download. I don't know. 

It should do a DHCP every time it reconnects. Download success will probably depend on how long the disconnection is for.

> 1. Is there an easy way to restart network discovery or enable wifi if the button is disabled?

Unplug/re-plug the Ethernet cable.  Get a compatible wifi dongle. I don't think the wifi button will ever go from an enabled to a disabled state on loss of ethernet connection, unless the wifi dongle is flakey or your power supply is not powerful enough for it to drop out. You could probably test with `lsusb` to see if the dongle has disappeared.

> 2. BONUS: How do you type Ctrl+Alt+F2 on a Mac in VNC?

It's not recognised over VNC - use SSH to get to the recovery console remotely.

Thanks for the VNC tip - I tend to use tightvnc instead, except when I want to VNC to the RealVNC server in Raspbian, which doesn't like it.

sarossell, 2018-03-21:
Thanks for the detailed response. Very helpful and much appreciated. I connect my iPhone to the Mac via USB and then share USB to Thunderbolt/LAN Bridge to the RPi, but your advice is still valid. I failed to mention the full details of my connection in my earlier post.

PINN is very impressive. You've really taken a good idea with NOOBS and made it much better.

I also prefer tinghtvnc, but sometimes it doesn't like me. ;@)

closed 2018-05-09 by procount