procount/pinn #136: Update PINN Support for Raspberry Pi 3 B+

tayrayb, 2018-03-22:
I tried using PINN on my new RPi, but it just gives me a rainbow screen with the voltage logo on the top right. I've used Arch Linux ARM perfectly fine along with the updated Raspbian on the RPi 3B+.

procount, 2018-03-22:
It's a work in progress....
You can try pinn-25b2.zip in the testing folder on sourceforge but some wifi drivers are missing. Please provide some feedback if you use it.

tayrayb, 2018-03-22:
Sure thing, I'll test it out now.

tayrayb, 2018-03-22:
I just booted it up, Wifi works perfectly fine, even picks up the 5Ghz networks I have.

milovanderlinden, 2018-03-23:
Works perfect on my Pi 3B + too. It is installing Archlinux ARM perfectly.

AmedeeBulle, 2018-03-23:
I used this 2.5b2 release to upgrade a project running on a 3B to a 3B+ (just kept my recovery.cmdline).
Worked perfectly.

closed 2018-03-25 by tayrayb