procount/pinn #137: Lakka & Retropie Error

hahagu, 2018-03-25:
Attempting to install Lakka or Retropie with PINN or NOOBS gives the following error.

```
Error downloading or extracting tarball
xz: (stdin): Compressed data is corrupt
Cannot write to '-' (No such file or directory).
./SYSTEM: Truncated tar archive
bsdtar: Error exit delayed from previous errors.
```

I am using a Samsung EVO 64gb (SDXC) microSD, formatted FAT32 with gparted utility.
I also tried using a 16gb microSD card to make sure that it wasn't the card's fault.

procount, 2018-03-25:
Which model of Pi are you using?
Which version of PINN are you using?

I have just tried installing Lakka2 on my pi 3B+ using v2.5.1 of PINN that I just released, and it installed without problem, although I don't think Lakka has been upgraded to use the new LAN and wifi of the 3B+ yet.

hahagu, 2018-03-27:
I am using RPi3
and is using the latest PINN version from sourceforge.

hahagu, 2018-03-27:
Hi, I just tried it with the latest version, 252a, and it still outputs the same error.
I am using the same board, with the same hardware.

procount, 2018-03-27:
Ok. I'll check the same combination. It could be you just have to wait til lakka is upgraded. Or just use an older version of pinn like 2.4.5f from the archive folder.

procount, 2018-03-27:
I have just installed Lakka_rpi2 with PINN v2.5.2a on a Pi3. PINN auto-upgraded the firmware and it installed from the internet and booted fine.

hahagu, 2018-03-28:
Hmm.. That's weird..
I wonder what's wrong..
Because it doesn't seem like I'm doing anything wrong..
btw, if it matters, I used gparted on ubuntu live to format the 64gb microSD and 16gb microSD as FAT32

procount, 2018-03-28:
I'm sure you haven't done anything wrong. ( I think I tried it again and I got a kernel panic :-/ )
There are some strange incompatibilities going on at the moment, due to the enforced firmware upgrade.
Please bear with me whilst they are sorted out.
Although your particular problem looks more like an interrupted download rather than "a failure to boot" problem.

If you want to try something....... 
Please download pinn-245f.zip from my archive folder on sourceforge.
Backup the bootcode.bin and recovery.elf files on your PINN 2.5.2a installed SD card.
Extract the 2 files: bootcode.bin and recovery.elf from pinn-245f.zip and replace the files on your PINN 2.5.2a SD card with these ones.
Try installing Lakka again.

If that fails with the same xz/bsdtar error again, replace all PINN files on your SD card with the files from pinn-245f.zip and try again.

Let me know how you get on.

procount, 2018-03-28:
How did they fail?

hahagu, 2018-03-29:
The same exact error.

procount, 2018-03-29:
So re-reading this thread, I think I assumed your problem was related to the recent problems with upgrading the firmware for the new 3B+ which seems to have had other consequences, although I hinted at an interrupted download problem. I think this is your real problem - for some reason your download is not working properly for Lakka.

To avoid other firmware problems, I suggest you use PINN_245f since you are using the RPI3.

I suggest you try downloading Lakka manually to a USB stick, insert the stick into your RPi and install it from there instead of directly from the internet. That way you have more control and can be sure the downloads have been done properly. Check each file size to make sure they are complete.

1. Go to http://mirror.lakka.tv/noobs/Lakka_RPi2/ and download all of the files and folders that you see at that location. 
2. Create a folder on your USB stick called /os/Lakka_RPi2 and copy this structure into the folder.
3. Insert your USB stick into your Pi and boot PINN. You do not need to be connected to the internet to install from USB.

(You could choose the "Download" option from within PINN, but as you _seem_ to be having trouble with this particular download, it may have the same problem)

hahagu, 2018-03-30:
This is weird, because I used the USB, and it gives
**`Error downloading or extracting tarball
xz: (stdin): Compressed data is corrupt
./SYSTEM: Truncated tar archive
bsdtar: Error exit delayed from previous errors.`**
Hmm... Maybe the Lakka tar itself is bad?

closed 2023-02-12 by hahagu