procount/pinn #138: Restrict OSes available to install to only those on USB drive?

dlabun, 2018-03-28:
Is there a way to restrict the list of OSes available to install to only those that are present on a USB flash drive? Essentially I don't want PINN trying to download an OS over the internet, but rather just copy the image from a USB drive to the SD Card. I am going to be deploying some RPi Compute Modules that will be connected to very restricted cellular connections.

procount, 2018-03-28:
Yes. It's in the instructions. See https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-install-oses-from-an-alternative-network-sources
So make sure you set `no_default_source` and delete any `alt_image_source`, `repo` or `repo_list` entries.

dlabun, 2018-03-28:
Excellent, Thanks for link. I'll give it a shot as soon as I'm done building a copy of PINN with my odd display settings.

closed 2018-03-28 by dlabun