procount/pinn #141: V.2.5.3 No Boot Menu After Installing Multiple OSes

d00di, 2018-04-04:
Greetings,
I installed four OSes and there is no menu at boot to pick an OS after they were installed. I'm only presented with the Shift option to edit PINN. All four OSes show as installed and only one boots automatically. Bug with 2.5.3?
Thx,

procount, 2018-04-04:
After you had installed the 4 oses and you were presented with the boot menu for the first time, did you tick any of the  sticky check boxes next to your oses?

d00di, 2018-04-05:
Hi. I don't recall being presented with a boot menu. So are you saying that there's only one opportunity to setup the boot menu after the OSes are first installed?
The only option I have now is to press Shift and then select Exit and then select the OS I wish to boot.
Thanks for the reply btw.

procount, 2018-04-05:
Have you read the documentation on the various ways PINN can boot an os:
https://github.com/procount/pinn/blob/master/README_PINN.md#booting-an-os-with-pinn

In particular, the following section on selecting a sticky OS to boot:
https://github.com/procount/pinn/blob/master/README_PINN.md#sticky-default-os

It sounds like when you first chose which OS you wanted to boot, you did so by ticking the box next to it which made it a "sticky" OS and would result in the behaviour you have stated. If you don't want this to happen, then go to the boot menu (by pressing shift and exit as you describe) and make sure no checkboxes are ticked. Just HIGHLIGHT the OS you want to boot and press the boot button, or press Enter.

Thereafter you will be presented with a boot menu of OSes to choose from each time you boot. If you don't choose within 10 secs, the last selected OS will be chosen automatically.

Ticking a checkbox to enable a sticky OS is a way of booting a selected OS more quickly because it avoids showing the boot menu, and will always choose that OS regardless of the last OS chosen (unless you interrupt the process as you are having to do now).

d00di, 2018-04-07:
That's probably what I did the first time. I verified that nothing was checked in the Sticky menu and then pressed "Escape" key rather than "Boot" and it rebooted and now a menu is presented. So if you're in the Maintenance menu then Exit and no Sticky default is checked and you select Boot it boots the highlighted OS but when you reboot from that OS you are still not presented with a Boot Menu. The key is press "Escape" after unchecking any sticky default... not obvious.

Thanks for your help. I had been using BerryBoot for a long time and wanted to try something else. PINN is good and I like the broader choices of OSes.

procount, 2018-04-11:
Esc reboots without setting an OS, so it effectively reboots back into PINN.  I don't think it is an essential step. Unticking the checkboxes first is the key to it working as you expect.

closed 2018-04-11 by procount