procount/pinn #143: v2.5.4 Build Failure (rtl8188eu patch)

dlabun, 2018-04-06:
I'm assuming this has something to do with the Realtek wifi drivers being removed in this release, but when building this version from scratch it will fail at patching the rtl8188 (see attached).

I do not need the wifi drivers at this time a temporary work around is good enough for me.

[error.txt](https://github.com/procount/pinn/files/1882447/error.txt)

procount, 2018-04-06:
Yes I think you are right. Not sure what Git did there.
Just delete/or rename buildroot/linux/linux-0003-8188eu.patch  and possibly buildroot/linux/linux-0004-rtl8812.patch. You may have to clean and rebuild linux if deleting the patches on their own does not work.

dlabun, 2018-04-06:
Yup, killing those two patches allowed a successful build.

closed 2018-04-27 by procount