procount/pinn #144: RetroPI

OxygenGS, 2018-04-06:
How do i install retroPI with pinn?

procount, 2018-04-06:
Did you just post this question on the forum, too?

Which RPi model do you have?
The current version 4.3 of Retropie does not yet support the RPi3B+.
There is a test version that apparently does, but I need to convert it to the NOOBS/PINN format first.

OxygenGS, 2018-04-06:
RPi3B+. Ok cool! When would that be?

procount, 2018-04-06:
Did you just post this question on the forum as hoenens? (no point in me answering the same question twice :wink: )

The rpi3B+ has limited OSes at the moment due to it being so new - maintainers have to update all their OSes to suit.

I have tried to download the test Retropie, but it failed twice with network errors. 
I'll get it done asap, but I'm very busy at the moment, so please bear with me.

OxygenGS, 2018-04-06:
Cool ty so much!

closed 2018-04-06 by OxygenGS

procount, 2018-04-07:
4.3.15 for 3B+ is now available as retropie3