procount/pinn #147: RetroPi 4.4 released for RPI B+

OxygenGS, 2018-04-15:
When will it be added? i did just update PINN to 2.5.5 but it still shows RetroPI 4.3

procount, 2018-04-15:
I already added 4.3.15 beta for 3b+ owners.
4.4 has only just been released. As I am away on a business trip for a week, you might have to wait a little while longer, I'm afraid.

OxygenGS, 2018-04-15:
Ow ok it's fine, ty :D

Denethor85, 2018-04-24:
Sorry, Retropie 4.4 is now avaiable? thanks

OxygenGS, 2018-04-24:
I didn't check yet.,

Denethor85, 2018-04-25:
I did not have time to try Pinn, but looking at the code there seems to be a link to Retropie 4.4 already

procount, 2018-04-25:
Yes, I have now replaced Retropie with the latest v4.4 Stretch version that also covers the 3B+.
PINN is updated to v2.6

OxygenGS, 2018-04-25:
Yes it works! The only thing now is: i did install Raspbian + RetroPI. How can i acces the retropi rom folder via Raspbian? Because i wanna download them via Raspbian because i cannot get FTP to work within RetroPI

procount, 2018-04-25:
This is a generic answer giving the principles (adjust as necessary).
You need to know which partitions each OS is mounted on. (HINT: Look in /settings/installed_os.json)
Make a mount point in Raspbian, e.g.:
`mkdir /tmp/roms`
Mount your retropie root partition where your roms are stored onto this mount point:
`sudo mount /dev/mmcblk0p# /tmp/roms` where you should replace # with the partition number of the retropie root partition (7 or 9 I guess, but see above)

Now you can download your roms in Raspbian and store them in /tmp/roms/(whatever_path_you_need)
(You may need to use `sudo` to copy the romfiles to /tmp/roms, or change the access rights/ownership of /tmp/roms/ to allow user pi to access them directly without needing sudo).

When you are finished:
`sync`
`sudo umount /tmp/roms`
and boot into retropie

OxygenGS, 2018-04-25:
That's like chinese to me, no idea how to do that haha.

procount, 2018-04-25:
Google is your friend!
An easier way - use a USB stick to transfer the roms from Raspbian to Retropie.

OxygenGS, 2018-04-25:
hmmmm
jeah i will do that its easier haha ty!!!!
But, how do i change things now so when if i in the future wanna add an OS but NOT wipe the current OSés?

procount, 2018-04-25:
Short answer - you can't.

PINN will expand each OS to fill up the SD card so there is no room for any new OS.
You need to install all OSes at once. 
Later, you may be able to replace an unwanted OS with another (if the partitions are big enough and compatible), but squashing the existing partitions to make space for new partitions to install another OS in - Not going to happen. 

Also see #148 and https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=142574&start=325#p1306413

So you could do it manually, but perhaps you'll have to learn a bit more before you attempt it. 😉

closed 2018-04-27 by procount