procount/pinn #148: Add/remove OS feature

gnadlr, 2018-04-24:
Just want to flesh out the idea from our recent discussion on the forum.

PINN could implement an add/remove OS feature without modifying the partitioning:

1/ Add an option (with quantity) for "Empty set of partitions" in OS install menu. If this option is selected, PINN will create an empty set of partitions (FAT+ext4) when running the installer.

2/ Add an option to install OS to an empty set of partitions.

3/ Add an option to remove an OS (making its partitions empty).

This is probably the most requested feature of NOOBS. I'm not sure if this request is better be made in the NOOBS repo, but since PINN recently implement an "reinstall" feature, I think this is just an expansion on that idea. You mentioned this feature is on somewhere on the road map, so I'd like to get any update on the news here.

procount, 2018-04-24:
See also https://github.com/raspberrypi/noobs/issues/488

procount, 2018-06-05:
v2.8.4 implements Project Spaces and the ability to replace an OS. I believe this satisfies this requirement.

closed 2018-06-05 by procount

gnadlr, 2018-06-06:
Glad to see it's implemented so fast. Cool!