procount/pinn #149: kali linux kernel panic

pctechadam, 2018-04-30:
Fresh install of kali linux from pinn fails to boot with a kernel panic error. On this SD card are three other working operating systems. I checked the cmdline.txt and it has the correct partition for the root file system.

pctechadam, 2018-04-30:
> 3b or 3b+?

No + just the 3b

procount, 2018-04-30:
My 3B is busy at the moment, but I tried installing Kali2 on a 2B from the internet and it booted fine to the login prompt. 
You could try re-installing just Kali to see if that helps. Otherwise I'll try and install it on a 3B when I get a changce to see if there are any problems with it, but it certainly looks ok on the 2B.

pctechadam, 2018-05-03:
I got the chance to test single install seems to work without issues.

procount, 2018-05-03:
What do you mean by a 'single install'?

- Using PINN to install just Kali2 on its own?
- Installing Kali2 as a normal image without using PINN?
- Using PINN to re-install Kali2

I hope to get my Pi3B back soon so I can test this.

procount, 2018-05-04:
Since the version of Kali that PINN installs predates the 3B+, it will only work with PINN when PINN has the old firmware installed. Did you try to boot it when the firmware button on the archival menu said "3B+" in red? If so, just click the button to downgrade PINN's firmware indicated by the button saying "0123" in blue, then try booting it again.

pctechadam, 2018-05-04:
Thank you. I had forgotten that i clicked on that. It works without any
issues thank for your help.

procount, 2018-05-04:
Cool!

This transitionary period is quite complicated....

closed 2018-05-04 by procount