procount/pinn #156: Auto re-install OS

procount, 2018-05-23:
The brief summary is:
1) add option/config to reserve fixed amount of space for partition p1
2) add a prioritized boot order before PINN takes user input of:
-- If exists os with overwrite config on USB stick, execute its installation, including overwrite of partitions as listed in its config file, and boot into the new OS; otherwise
-- if exists an OS with overwrite config set in the /os folder on p1, execute its installation, including overwrite of partitions as listed in its config file; clear overwrite; and boot into the new os; otherwise,
-- do what pinn normally does, i.e. check for shift key.. etc, which will boot the default OS. 

 noafterglow

procount, 2018-05-23:
The ability to reserve space on P1 is now done,

noafterglow, 2018-06-12:
Awesome.  I've played with PINN a bit.  reserve=xxx works great.  

It would be handy to have silentinstall work from the USB drive too, or perhaps to have a separate option for that?

procount, 2018-06-12:
I've not touched silentinstall at all. Surprised it doesn't work when pinn is installed from usb. Or do you mean pinn is on sdcard but you want the os to be silently installed from usb to sd card?

noafterglow, 2018-06-12:
The second one.  Pinn on sd, os directory on usb... ideally it would be
silent overwrite and not install...

I can't figure out how that even happens.  I See where it figures out the
command line options and sets the bool variable but cant figure out where
that gets used....

procount, 2018-06-13:
Are you sure siletInstall will not work with with PINN installed on SD card and OS installation files on a USB stick?

I haven't tried it myself, but looking at the code it looks like it should work. However, take care of the following points:
1) silentinstall option must be specified.
2) There should be no OSes already installed.
3) There should be only 1 OS available for installation (1 OS on USB stick, none on the SD card and none accessible via the internet)

What happens if you try this scenario?

noafterglow, 2018-06-13:
It boots, reformats the card as specified, and then brings up the the OS
installation screen, with just the one OS available.   You have to check
the box, and click the button to make it do the install, which then
completes properly.

If I subsequently copy that os to the SD card and remove the USB stick, it
properly installs the OS automatically.

the recovery.cmd had the following options added:  runinstaller
silentinstall reserve=1300 vncshare no_default_source

Ihor Lys
617-470-2740

procount, 2018-06-20:
I'm thinking that I could implement some sort of 'Auto-Reinstall' option.
If PINN finds an OS installation with a specific flag in it's os.json file, such as 
`"auto-reinstall": true`
Then it will check the version/date of that OS, and if newer than the one installed, it will automatically re-install it. I think the installed OS will also have to have that flag set to prevent inadvertent overwriting.

I have a potential issue that with the advent of ProjectSpaces, I could have multiple OSes with the same name installed. So to prevent overwriting/re-installing all of them, I thought an option would be to replace the `true` value with the boot partition number of the OS to be re-installed.
How does that sound?

noafterglow, 2018-06-21:
That could certainly work.  I think if you have to have the flag in the
os.json file, then you should be good enough.  I'm not exactly sure of the
value of project spaces having the same os installed multiple times...
wouldn't you rename them?
Ihor Lys
617-470-2740

procount, 2018-07-02:
Getting half way there with the new `select=` option in v2.8.5.

procount, 2018-07-06:
> It would be handy to have silentinstall work from the USB drive too, or perhaps to have a separate option for that?

I haven't tried this test specifically, but I think if you add `silentinstall select="Raspbian,waitusb" disbalesdimages` that it will install Raspbian from USB. It will only work if there are no OSes already installed. Change "Raspbian" for the name of your OS. The `disablesdimages` is just to make sure it doesn't install a version from the SD card  instead. You may also want to block remote downloads as well.

procount, 2018-07-09:
I have added a new option called `silentreinstallnewer` in v2.8.5.1 which almost works like you wanted, but does it slightly differently. Since it requires the recovery menu to be entered to detect all OSes, you may need to use the `forcetrigger` and `remotetimeout` options for it work unattended. It may delay boot up time a little. Let me know what you think.  Full details will be in README_PINN.md.

noafterglow, 2018-07-09:
Great.  I will look at it today!

procount, 2018-07-13:
Is it ok for you?