procount/pinn #159: 【Feature Request】Add "reboot" button on the bootmenu

qcsunny, 2018-06-08:
I set "remotetimeout=5 forcetrigger vncinstall bootmenutimeout=5" in the recovery.cmdline.

However,when I finished installing "project space" , it would automatically change into bootmenu, there was no system to boot obviously. I had to power off the rpi manually. 

So I guess, is it more convenient while a “reboot” button next “boot” button？

[image]

procount, 2018-06-08:
Ah yes. I haven't documented it but you can just press Esc and it will reboot back into PINN. It probably shouldn't go to the boot menu if there are no bootable OSes installed anyway.

Do you find remoteboottimeout=5 is long enough? I think I had to set mine to 20 to allow me time to get in.

qcsunny, 2018-06-09:
Thanks for your reminding.
1. Nope, it seems like not work for me.When I only install `project space`，then it shows this picture and I press `Esc`, it still go to the bootmenu
[image]

2. `remotetimeout=5` may be not enough because I have failed to get in the PINN menu several times. When I get the remote pictures , it has already entered the boot menu. So I would like to reboot it immediately and try it again, instead of booting the system first. 

Normally, I will get the following picture if I connect the pi at the right time (Network ACT led light up or wait for seconds？)
[image]

`boottimeout=5` is enough for me. Once I get the remote pictures, the countdown will stop and let me to select the OS.

qcsunny, 2018-06-09:
Sorry, I misunderstood what you mean.
Now, when I press ESC on the boot menu, it will restart.

qcsunny, 2018-06-09:
In general, I can press ESC on the bootmenu to let it reboot, “reboot” button is no need. :smile: I don't notice this feature before.

And, it cannot detect wether the real  OS is already installed and enter the bootmenu. Is the situation  just for me?

procount, 2018-06-09:
I have a few things to fix, but you have a workaround, so they will be low priority.
1. document pressing Esc on the bootmenu will reboot into PINN.
2. Add a reboot button to make this more explicit (especially for CEC users)
3. Prevent going to bootmenu after installing OSes, if there are no bootable OSes installed.

procount, 2018-07-02:
Done in v2.8.5

closed 2018-07-02 by procount