procount/pinn #160: HDMI-CEC does not work (pinn 2.8.4 on Raspberry pi 3 B+)

pgScorpio, 2018-06-11:
I CAN use "any key" on samsung tv remote to enter the pinn menu, but from there on the tv remote does nothing.
tail -f /tmp/debug does not show any cec activity at all. 
also when booting into libreelec, NO tv remote.

A plain libreelec install however DOES work with the tv remote.

procount, 2018-06-11:
Last time I tested on my Samsung, it worked fine. But I can't test again for another week. In the meantime it might be useful if you could test if v2.8 or earlier (before programmable CEC was added) works on your tv.

procount, 2018-06-12:
Not automatically. It tells you if a new version is available. It's up to you whether you upgrade it or not.

pgScorpio, 2018-06-12:
I did some quick testing, but the problem seems to be related to the TV (Samsung UE32ES6100) somehow.

I tested a new PINN 2.8 card and installed LibreElec, same result..., no CEC control.
Than I booted from the SD card with the LibreElec image installed (No NOOBS, PIN, BerryBoot) which was working before. But now CEC was not working anymore...

So I checked/changed cabling/inputs etc, no change...
But after unplugging the power of the tv for a few secs  it worked again !
Than I tried PINN 2.8 card again, CEC working !
Than I tried PINN 2.8.4 card again, CEC working !

Than I reformatted the PINN 2.8 card and reinstalled PINN 2.8.4 and LibreElec, CEC NOT working !
Unplugged the tv again, CEC working !

So the installation of a new PINN sd card seems to mess up my TV's CEC, which only recovers after removing power from the TV. 

(But I still do not understand why entering the PINN configuration by CEC still works while the other CEC functions do not work anymore. Maybe the tv starts sending invalid CEC messages which will trigger the config menu but are ignored for anything else ? Remember that /tmp/debug does not show any CEC activity when CEC is not working. But it does after powering off/on the tv and CEC is working again.)

So SAMSUNG tv owners with not working CEC: Try removing power from your tv ! 
(Unplug !, Power switch and/or remote will not work !)

procount, 2018-06-12:
Thanks for sharing. That is very strange behaviour.