procount/pinn #161: Pi3b - Pi3b+ pinn 2.8.3 - 2.8.4 w/ ext usb issues

cognitiaclaeves, 2018-06-13:
I got a shiny new Pi3b+ today. The first thing I did was attempt to swap out the SD card from my new Pi3b and boot it. I got a lightning bolt -- an indication that the software (pinn) didn't have the right firmware? for the 3b+.

I had installed everything on an external USB, so I figured that I could probably recover from reformatting the SD card and reinstalling pinn easily enough, but this really wasn't the case.

Though I was able to boot pinn after making this change.

Right now I'm installing 7 'project spaces', leaving room for an eighth, which I hope I'll be able to coerce into a project space that references the OS on my USB. As I'm thinking about this, I left the eighth unchecked, as I didn't want to overwrite my previous attempts at getting the old OS working, but I observed that the last time I did this, it deleted the old partitions, so this probably means that it will delete the eighth, and I'll need to cannibalize one of the OSes that I installed.

I've saved all the files from the original USB, so I should be able to copy / reference anything needed to set this up.

Is there a way to get an old OS running on a USB stick in a newly formatted SD card with pinn installed?

procount, 2018-06-13:
It would have been better to:
1) Whilst in the 3B, Upgrade PINN to the latest version
2) Still in the 3B, use the menu option to change PINN's firmware for the 3B+ version
3) Maybe update each installed OS individually to ensure they will work in the 3B+.
before moving the SD card across to the 3B+. (This is similar to what you would need to do with a plain Raspbian installation as well).

> I had installed everything on an external USB

I guess you mean the "rootfs" of every OS. PINN and the boot partitions of each OS would still have been on the SD card, so reformatting it will have removed all your boot partitions.

It's not really possible to "leave room for an 8th project space" if you only install 7. Project Spaces will divide ALL of the available disk space between them, so if you need 8 OSes in the future, best to install them all at once. That is one of the reasons for introducing Project Spaces. You can replace an OS, but not easily ADD an OS without doing some serious lobotomy of the disk space partitions and other files. In any case, Project Spaces are not really relevent to fixing your problem here.

> Is there a way to get an old OS running on a USB stick in a newly formatted SD card with pinn installed?

Maybe, but it's not going to be easy, and you still may have to do some serious lobotomy of some installation files.

I am away from any means of testing this, so are are largely on your own, but this is how I would proceed in principle. This is not a step-by-step guide, and there may be some gaps that you will need to work out how to do yourself.

Essentially, you still have your rootfs partitions on your USB stick, but you have no boot partitions on your SD card. 
So you need to restore PINN and the boot partitions on your SD card, and then link them up again. This is quite advanced stuff if you ar ea linux newbie.

So, start with a fresh SD card and install PINN on it, and get a fresh USB stick that you will use temporarily.
Once PINN is booted, install all the same OSes that you had before onto your new USB stick. This will restore the boot partitions on your SD card. Hopefully they will be in the same order as before, but maybe not.
Your new USB stick is now redundant and you can reformat it or do what you want with it.

You now need to link up your new boot partitions on the SD card with the old rootfs partitions on the old USB stick.
So boot your new PINN sd card with your old USB stick attached. (More than likely none of the OSes will boot).
You can now drop into the recovery shell with ctrl-alt-f2 and login with username=root, password=raspberry.

You need to edit the /settings/installed_os.json file. The only editor you have available is 'vi'. If you don't know how to use this and want to use a more familiar editor, you will need to copy this file somewhere else so you can edit it elsewhere, then copy it back again.

In settings.json, each OS is listed with the partitions where the boot (1st entry) and root (2nd entry) partitions are located. 
The rootfs will be using a PARTUUID=XXXXX-PP reference that will refer to the NEW Usb stick that I told you to reformat. You need to change this to point to your OLD USB stick. (The XXXXX part should be the same for all installed OSes - it refers to the USB stick. THe PP part refers to the partition number on the UBS stick.
2 ways: 
1) Find out the PARTUUID of your old USB stick and substitute the XXXXX part of the PARTUUID value (Use sudo fdisk /dev/sda , x for extra functionality, i to view & change disk identified, ctrl-c to cancel)
2) if you only insert 1 USB stick at a time, you could use the old way of referencing the partitions by substituting PARTUUID=XXXXX-PP with /dev/sdaPP (delete any leading 0's for PP < 10)

Then you need to go through the cmdline.txt file on each boot partition of each OS and do the same modification. (There is a Fix feature on PINN's Archive menu. It has a 're-run partitions_setup.sh' option which may do this last step for you once the installed_os.json file has been updated. It may even work better as it will also update other files, but I'm not able to test it for this specific use case).

Hope this helps, and good luck! (I think you will have learned not to do this again!)

cognitiaclaeves, 2018-06-13:
Thank you so much for your quick response.

I'm not new to Linux, so everything you said seems well within reach.  I also became pretty familiar with some of the files you mentioned last night, when I was trying a number of different approaches to getting it working again.

I figured you were going to mention the firmware toggle. I found that switch ... shortly after I reformatted the SD card ...

In my old config, I noticed that it used PARTUUID for the old USB stick. I'm not positive, but I think I should just be able to reuse that ... I'll give it a shot.

procount, 2018-06-13:
Yes, if you kept a backup of any of your /settings files or /boot files you could get it from there. Or look in the /etc/fstab file of each os partition on your old usb stick.

cognitiaclaeves, 2018-06-13:
I'm only trying to recover one instance that uses the whole usb. When I first started playing with the pi, I only installed that one OS, thinking I could add others later. I quickly found out that wasn't the case.

Now, I'd like to install differently, so I have the setup I mentioned -- 7 spaces + a number of other OSes that I downloaded and installed at the same time. One of them is an instance of retropie. I can run it as long as the root partition is pointed to what was installed /dev/mccblk0p15, but when I change it to the PARTUUID (in /settings/installed_os.json and using PINN's (e)dit config option for the OS), it seems unable to load the kernel.

Is there a connection somewhere that I've/we've missed?

procount, 2018-06-14:
When PINN installs an OS, it runs a partition_setup.sh file to fix up all the references to its new location. Have a look at the partition_setup.sh file for Retropie to see what files it needs to fix up.

Some OSes will not run from a USB because they don't all support the PARTUUID syntax. Check in the os.json file if it supports_usb_root/boot.

cognitiaclaeves, 2018-06-15:
Giving up on this and starting over. Thanks for your help in trying to figure it out!

closed 2018-06-15 by cognitiaclaeves