procount/pinn #163: vncshare fails sometimes

noafterglow, 2018-06-21:
not sure exactly why.  ssh continues to be available, but vncshare returns "Connection refused"

this happened on about 3/4 of the pi's in one installation.  Could be related to network timeouts.

I have not been able to reproduce this in my lab.

procount, 2018-06-21:
Does "sometimes" mean fails intermittently on a single Pi, or fails permanently on some Pi but works on others?
If the latter - just double check you have vncshare in recovery.cmdline. I assume you have already, but the only time my VNC didn't work is when I thought I had added it , but hadn't.

noafterglow, 2018-06-21:
no, its truly random.  We had 16 units all identically programmed, and
could only get to 3 of them through ssh.  Had to copy the os folder to the
card, then add reinstall and silentinstall to bring them back up.

Ihor Lys
617-470-2740

procount, 2018-06-21:
Are these units all headless as you are using VNC/SSH?
VNC is not a separate program, but it is integrated into the PINN recovery program itself using QT's built-in features. So the only other thing I could think of is if PINN had crashed for some reason. If you are headless you may not notice. You could probably tell if you did `ps` or `tail /tmp/debug` might indicate a problem.

noafterglow, 2018-06-21:
Interesting...  Yes.  they are headless.....   so I don't get to see
anything.....

so is it simply a matter of re-starting PINN?

Ihor Lys
617-470-2740

procount, 2018-06-21:
It could be, if that is indeed the problem. Best to just reboot it (from ssh: `echo b>/proc/sysreq-trigger`)
But then the question is why did it crash? If it happens again, look at /tmp/debug.

noafterglow, 2018-06-22:
Reboots did not bring them back.  That was the first thing I tried.  I'm doing two more batches today, I will let you know what I find.  I did however write a python script to automate remote install and upgrade using pinn, and some serious abuse of ssh, so I technically don't need graphics or USB for it this time....

noafterglow, 2018-06-24:
I finished upgrading remotely 100 units to run PINN 2.8.4, rebooted, and then scp an os folder 
 into the p1 partition remotely, then added reinstall and silentinstall to the command line, and rebooted again.  PINN happily installed the os as it should have.  There were no issues.  

So, the only difference between this and what happened to create the vncshare issue is that the os was on the USB before, and not on /dev/mmcblk0p1.  I was able to also remember that when I had the issue, PINN still changed the size of p1 as requested, so it did not "crash" if that is what was indeed happening until **after** that part of the process.  My recovery procedure before was to ssh while vncshare was not available, copy the folder from usb over to p1 manually, add back silentinstall and reinstall, and reboot.

So all of this points to some issue with USB support.

procount, 2018-06-24:
Glad you sorted it.
One other thought - IIRC, if you are installing PINN for the first time on a USB stick when the runinstaller option is present, VNC is not available until after PINN has repartitioned the drive. On a large USB stick, this may take a while. So under these conditions, ssh is available but not VNC.

noafterglow, 2018-06-24:
That's definitely not it... I'm returning to the location where I saw this
next week, and I will deliberately try to recreate the exact situation.

lurch, 2018-06-26:
If you're managing so many RPis at once, perhaps https://etcher.io/pro/ or https://resin.io/ or https://resinos.io/ or http://pinet.org.uk/ would be better-suited? :man_shrugging:

closed 2023-03-25 by procount