procount/pinn #164: HDMI-CEC doesn't work

bobblkabb, 2018-06-26:
Got a dual boot setup with libreelec and retropie on a RPi3. Systems are installed on an external SSD.
CEC works on libreelec but during pinn (v2.8.3) boot it doesn't react. Also tried _tail -f /tmp/debug_ but no reaction on any remote control button. Got a LG tv.
What can I do? Is there maybe a way to copy CEC settings from libreelec?

procount, 2018-06-26:
libreELEC uses a different CEC library to PINN that is probably more robust, but I don't use it for disk space and licensing reasons (I may have to review that library decision ). Some other users have had success by changing the configuration of their TV, or even just powering it off prior to using PINN.

Maybe LG TVs need some different initialisation? It's a bit difficult to debug without an LG TV at hand. If you are willing to do some debugging on it I may be able to fix it. There is some software called cec_client that allows you to send specific commands to your TV. You might like to try that to see if it shows you some specific commands that need to be sent. See https://blog.gordonturner.com/2016/12/14/using-cec-client-on-a-raspberry-pi/ for example. I can also output additional debugging information from PINN so you can see a bit more about what is happening on the CEC interface, if you want to go down that route.

bobblkabb, 2018-06-26:
Need the libreelec CEC output? Or did you mean from pinn but there's no cec-client..

echo "scan" | cec-client RPI -s -d 1:
```
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        LG
osd string:    TV
CEC version:   1.3a
power status:  on
language:      eng

device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        LG
osd string:    CECTester
CEC version:   1.3a
power status:  on
language:      eng

currently active source: unknown (-1)
```

What info do you need?

procount, 2018-06-26:
I'm not sure what CEC output LibreELEC provides, but the more info you can provide, the better!

On PINN, at the top of /tmp/debug, can you just check that the messages:
**CecListener thread started**, and
**CecListener done initializing**
appear, and that there are no other messages between them? Other than that, I don't actually think there is much more debugging info I can provide that isn't there already.

I intended for you to run cec-client on another os, like Raspbian, which I guess is what you have done. I don't have any specific tests for you to do at the moment. I'll have to try it on my Samsung TV and see what is necessary, then maybe we can compare results.

There's one additional fix I just found that you could try.
On the PINN recovery partition, (where you will find `recovery.cmdline` create a new file called `config.txt` file and add the line `hdmi_force_hotplug=1`. It may not do anything, but worth a try.

bobblkabb, 2018-06-26:
tail -f /tmp/debug:
checking size of file: "https://datapacket.dl.sourceforge.net/project/pinn/os/ProjectSpace/prfboot.tar.xz"
checking size of file: "https://datapacket.dl.sourceforge.net/project/pinn/os/ProjectSpace/prfboot.tar.xz"
Length: 208 files lieft:   6  "ProjectSpace"
Length: 208 files lieft:   6  "ProjectSpace"
Length: 208 files lieft:   6  "ProjectSpace"

procount, 2018-06-26:
`Tail` only gives you the end of the file. You need to use `head` or `less` ti see the beginning.

bobblkabb, 2018-06-26:
head /tmp/debug:
```
Keyboard found: "/dev/input/event0"
2.8.4
Board revision is 10494082
CecListener thread started
CecListener done initializing
Found recovery.rfs at "mmcblk0p1"
PINN drive: "/dev/mmcblk0"
Loading CEC mappings from ":/cec_keys.json"
Shift detected
Starting dbus
```

procount, 2018-06-27:
Well, that seems ok. I'll try doing some digging to see if I can identify why LG  doesn't work. But CEC is implemented differently by each manufacturer so it's quite difficult. I'm afraid it might take a while.

bobblkabb, 2018-06-27:
They call their cec implementation _simplink_. Just chattet with the hotline but didn't get any further information.

bobblkabb, 2018-06-29:
[cec-output.txt](https://github.com/procount/pinn/files/2148015/cec-output.txt)
Here's the output from my libreelec system.

gauner1986, 2021-12-07:
Any updates on this? I'm also having this issue as i'm using an LG TV. Willing to debug as well.

MindTooth, 2022-02-12:
> Any updates on this? I'm also having this issue as i'm using an LG TV. Willing to debug as well.

Same.  If there is anything that I can do to help make LG TVs work, I'll gladly do so.  Just need a nudge in the right direction.

procount, 2022-02-14:
CEC is a bit of a mystery to me, I'm afraid, not helped by the fact that each manufacturer implements it differently. There is a library called Lib_cec that tries to hide all the vaguaries of each manufacturer, but I don't use it because of licensing and space reasons. Instead I have followed Berryboot and use the same inbuilt RPF libraries. It works fine on Samsungs but I've not been able to test it on anything else.

A couple of things you could do is to 1) try the latest Berryboot to see if CEC works with your LG TV. If it does I can probably fix PINN to match. And 2) Try experimenting with the cec_client on a Raspberry Pi OS installation to see if you get same or different results to the above posts.

Failing that I may have to write some specific programs to debug the interface, or add more debugging to PINN to see what is going on.

MindTooth, 2022-05-29:
You know what, it does seem to be working now (latest release, not sure if that's relevant).  The mouse work, yet very slow, and I can press the red button to tab through the top menu.  Also, the back and enter (OK) button seems to work okay.

The TV adds profiles for various connected devices.  I suspect that the one added automatically did not correspond to a "clean" CEC device.

Not sure if there is a place where I can see all current mappings?  Would be nice to see if e.g. green and yellow could be arrow up and down.  Or perhaps the number keys can also do actions.