procount/pinn #167: Reinstall fails with "Partition not big enough for new image" at 2nd reinstall - still

bobblkabb, 2018-06-30:
Seems #152 isn't resolved (v 2.8.3).
Got 2 distros on an external SSD. Used gparted to resize partitions because retropie doesn't need much space but libreelec does.
After that retropie didn't boot anymore. Boot process stopped right after those raspberries are shown. There's no text prompt. So I tried to reinstall retropie and I was getting that error.

What can I do? How does this workaround in #152 work? What exactly do I have to rename?

procount, 2018-06-30:
Maybe gparted messed it up?
If you don't mind starting again, a better way is to use Matt's webapp at http://pinn.mjh.nz to set your partition sizes as you want.
Re: #152, the referenced file is /settings/installed_os.json (where settings is mounted on the 5th partition - /dev/mmcblk0p5 or /dev/sda5). You could paste the contents here.

bobblkabb, 2018-06-30:
Think the settings partition is empty. Can't look right now because I'm using libreelec right now but I'm pretty confident that it was a blank parition last time I looked. Was wondering.
So there should be a installed_os.json like in #152 . Is that everything?

procount, 2018-06-30:
If Pinn knows you have retropie and libreelec installed, then it is not empty. You may be looking in the wrong place. There are other files there as well

bobblkabb, 2018-07-01:
My bad! I got confused with a partition called 'settings'. 
That webapp for partitioning looks complicated and I already have a fresh installation. 
Before next resizing I will make a clone..

closed 2018-07-01 by bobblkabb