procount/pinn #173: Only boot on HDMI - Detection problem?

P1dum, 2018-07-17:
Hi,

After some weeks perfectly working, my boot screens selection started going mad. 

Typical usage : 1280*1024 and 1920*1080 screen, both DVI wires with hdmi adapters (HDMI only wires used for debugging, without improving any result).
And when I need, it starts only on the offcial 7" sensitive screen, permanently connected.

I use my RPi for dev small softwares in C using ncurses libs, far away from playing with deep RPi config files.

But, a day, everything went wrong : I was unable to boot the RPi on any screen in HDMI. After some changes in the configs files, like including hdmi_ignore_edid=0xa5000080 into the config.txt in the root partition, I could display again, but on the 1080 screen only.

I can now boot on the screen I want for the first part of the boot (the PINN screen), but after it seems to boot only on the HDMI output, whatever I write in the /config.txt or /boot/config.hdmi-dsi files.

Have you an idea on what's going on?

Thanks.

procount, 2018-07-17:
Errr, no idea.
Have you updated PINN recently?
What version are you using?
Are you using the DSI command line option?

Please post the output of your:

- recovery.cmdline
- config.txt on PINN partition (if any)
- config.txt/hdmi/dsi on the boot partition

Maybe it's just a typo, but:

>  like including hdmi_ignore_edid=0xa5000080 into the config.txt in the root partition, 

..... It should be the boot partition. :wink:

P1dum, 2018-07-17:
Thank you for your fast answer !

When I spoke about the root partition, I meant the partition /dev/mmcblk0p1, the boot one, yes.
I got PINN V2.8.4.

So here are my files : 
recovery.cmdline (from the place I mounted mmcblk0p1): 
quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline repo_list=http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json dsi (in one line)

config.dsi ( at /boot/ in the raspbian partition), the uncommented lines:
hdmi_ignore_edid=0xa5000080
lcd_rotate=2

config.hdmi same as .dsi : 
ignore_lcd=1

overscan_left=-110
overscan_right=-110
overscan_top=-110
overscan_bottom=-110
//1080p compatible

config_hdmi_boost=9 //Avoiding some pixels blinking

hdmi_ignore_edid=0xa5000080

I will try to comment the overscans to see if I can have the displaying back on my 1280x1024 screen.

Thanks !

P1dum, 2018-07-17:
Up : Nope, the screen keeps his deep sleep state :/

I repeat, before this, passing from one screen to the other worked OK, the desktop fitted perfectly with the resolution, automatically.

procount, 2018-07-17:
I'm not sure why you use hdmi_ignore_edid, then? You are preventing the RPi from identifying the type, size and resolution of the monitor, so it cannot auto-adjust.

If you need to set up your HDMI monitors, then the RPI needs to identify them. You can also use conditional filters in config.txt based on the monitor it detects to set different configurations. See https://www.raspberrypi.org/documentation/configuration/config-txt/conditional.md

Have you looked at the example config files for DSI in https://sourceforge.net/projects/pinn/files/os/Retropie2/dsi/ ?

P1dum, 2018-07-17:
I don't know why I use the hdmi_ignore_edid, it is the results of some tries to have a correct displaying back, and it works for my 1080 screen.

I tried the .dsi config file, and had not any effect.

Thanks for your help.

P1dum, 2018-07-18:
Ok so after a battery of tests it appears that:
- if I keep in config.txt (boot) hdmi_ignore_edid=0xa5000080, it will boot on HDMI regardless if there is a wire on the HDMI output
- if I don't, it will boot on dsi regardless if there is a wire on the HDMI output.

Each change works sucessfully on the boot screen, but when the reboot occurs on the desired OS, I have the behavior above.

Can you tell me how do you test if something is connected on the HDMI output?

Thanks

procount, 2018-07-18:
Have you read: https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-auto-switch-hdmidsi-screen-configurations

You do realise that there are TWO boot partitions with TWO config.txt files?
There is one for PINN and each OS installed has its own config.txt file as well.

When the RPI starts, if a DSI screen is connected, the HDMI is ignored - it is not possible to identify if it is there or not. So to reverse the priority of the HDMI/DSI screen (in your OS) when PINN boots, its config.txt file is set to disable the LCD screen so that the RPi can detect the HDMI screen. (This is done with `tvservice -n`)

In your OS boot partition (probably /dev/mmcblk0p6, normally labelled `boot`) you need to create `config.hdmi` and `config.dsi` If PINN detects a HDMI monitor, it will copy config.hdmi to config.txt (in the boot partition of your OS) to disable the LCD and enable the HDMI. If not, it will copy config.dsi to config.txt (in the boot partition of your OS) to disable the HDMI and enable the DSI screen. So now you have figured out how to switch between the 2 screens at startup, setup your config.hdmi and config.dsi files to select their appropriate screens, and make sure the DSI screen is disabled in PINN's config.txt file, otherwise it will never detect the HDMI screen. 

This means there is a limitation of only using PINN on the HDMI screen. If you try to use PINN on the DSI screen, it will never detect the HDMI screen, so it will always boot your OS on the DSI screen.

P1dum, 2018-07-23:
Thank you for your answer and sorry for the confusion between the config.txt files when describing my files.
I forgot that I needed to display the PINN boot screen on the HDMI output.

So, when I try to boot on the 1280*1024 screen, it doesn't even wake up. But, when I try to boot on the 1080p screen, the screen shows that the input resolution in 1280*1024 (with hdmi_ignore_edid commented on each config files).

On the PINN root partition (/dev/mmcblk0p1), I constantly keep ignore_ldc=1 and I tried some cases with and without hdmi_ignore_edid=0xa5000080.

There are the screens: 
Speechi 1080p
Belinea 1280x1024

The test matrix:
a : /dev/mmcblk0p1 config.txt
b:  /dev/mmcblk0p6/ (OS)/boot/config.hdmi

So:
Speechi : a = not hdmi_ignore_edid   b = not hdmi_ignore_edid => Boot on DSI
Speechi : a = hdmi_ignore_edid   b = not hdmi_ignore_edid => Boot on HDMI but with a terrible resolution
Speechi : a = not hdmi_ignore_edid   b = hdmi_ignore_edid => Boot on DSI
Speechi : a = hdmi_ignore_edid   b = hdmi_ignore_edid => Boot on HDMI, just need a bit of negative upscalign.

Belinea : a = not hdmi_ignore_edid   b = not hdmi_ignore_edid => Boot on DSI
Belinea : a = hdmi_ignore_edid   b = not hdmi_ignore_edid => Can't display on any screen
Belinea : a = not hdmi_ignore_edid   b = hdmi_ignore_edid => Boot on DSI
Belinea : a = hdmi_ignore_edid   b = hdmi_ignore_edid => Can't display on any screen

Whatever the screen I achieved to boot, when I try to obtain some informations with tvservice options, I can't. I have : No device connected, 0 DMT and CEA modes found etc...

Thanks for your help and your time.

P1dum, 2018-07-24:
Ok I tested with another RPi 3 and it works well for both screens :/
And I tested the first RPi only with HDMI (without the sensitive screen plugged), didn't work too. 

Same weird behaviour as on this thread:
https://www.raspberrypi.org/forums/viewtopic.php?t=47095

Maybe the quality of the HDMI port embedded on the Pi sucks ...

I'll continue to try with the new one to see if the default appears again.

Thanks

closed 2018-10-04 by procount