procount/pinn #174: Question: Can PINN be used for atomic updates of an OS?

mortenlj, 2018-07-22:
I want to run HypriotOS on my Raspberry Pis, but I would like to be able to do automatic atomic updates of the OS in the same way that you can do with CoreOS on x86 systems.

My plan at the moment looks like this:
- On first boot, use PINN to install latest version of HypriotOS, create a secondary OS partition for future use (a Project Space) and a data partition for data that needs to be preserved when upgrading the OS.
- When it's time to update, install the new version of HypriotOS on the secondary partition, switch to booting from that partition on next boot and do a reboot when convenient
- Repeat every time a new version of HypriotOS comes out, switching partitions back and forth

All of this in a headless setup, so fully automated, not doing any manual selection in any menus. The initial setup on first boot should be done by creating an image with a modified `recovery.cmdline` and any needed additional files.

There are a few questions:
- Is it possible to install an OS using the PINN framework from a running OS, or do I need to reboot into PINN, and then install to the secondary partition?
- Ideally, I would have two partitions that fit HypriotOS, and the rest of the disk for the data partition, how would I do that, if at all possible?

But, most importantly, is this at all feasible to do?

(If there is a better place to ask this question, feel free to point me in the right direction)

procount, 2018-07-22:
All feasible. But....
> so fully automated, not doing any manual selection in any menus. 

You'll need to do some work to make it all automated with your own scripts.

You could install the other os from the first os too. But detecting when a new os is available and upgrading the right pattition would need your own careful scripts.

mortenlj, 2018-07-23:
I write software for a living, so writing some scripts for this isn't a problem. I also have no intention of running anything critical on this setup in the foreseeable future, so if I have some catastrophic failures in the beginning it's only going to cost me time.

I'm going to continue working in this setup then. I might even contribute some examples or docs if/when I get it working.

Any tips on how to best create the partitions like I want them?

procount, 2018-07-23:
Sorry for being brief. I will have more time to assist in a week, but don't let that stop you asking me for further help :)

1. Determine the size of the partitions you want.
2. Setup 2 project spaces and 1 data partition using http://pinn.mjh.nz of your required sizes.
3. After reboot, replace 1 or 2 project spaces with hypriotos
4. Look at PINN source for how I detect new OSes available (silentreinstallnewer or pinn_update)
5. Installing an os is just xz | untar to appropriate partitions.
6. You'll have to run partition_setup.sh after each reinstall. Adjust it to map your data partition at the same time.

mortenlj, 2018-07-23:
No worries. I fully understand that open source projects aren't always the most important thing to spend time on. You've given me enough to work with so far.

lurch, 2018-07-23:
Sounds like you're re-inventing a lot of what https://resinos.io/ and https://resin.io/ already does? :man_shrugging:

mortenlj, 2018-07-24:
> Sounds like you're re-inventing a lot of what https://resinos.io/ and https://resin.io/ already does? man_shrugging

I've looked at resin, even got so far as installing on two pis before I realised that the only published build was a dev build, which has useful features for testing, but not for actual use. For instance it has passwordless root login enabled on port 22222, which can't be disabled. There were a few other issues I can't remember right now, but the root login was the deal breaker for me. It would be possible to fix by doing a custom build, or signing up for the managed service, neither of which is something I want to do.

lurch, 2018-07-24:
ping @shaunmulligan and @alisondavis17 and @agherzan in case this is useful feedback for them... :slightly_smiling_face:

mortenlj, 2018-07-26:
I've been reading up on the docs and trying to plan out how this would work. I figure I need to hook into the process at various places so I've been looking for available hooks to use. Just to confirm my understanding, is this list correct?

- `/background.sh`
-- Executed in the background during installation
-- Will this be killed at the end of the installation, or will PINN wait for it to exit?

- `/pinn_init.sh`
-- Executed before PINN starts

- `/before-reboot.sh`
-- Executed before booting an OS
-- Is this always executed, or just after installing a new OS?

- `partition_setup.sh`
-- Executed before or after installing a new OS?
- Part of the installed OS, not on PINN partition?

- noobsconfig
-- Not a script that is executed
-- Unpack an archive over the newly installed OS
-- Before booting into the OS, or as part of the first start inside the OS?

procount, 2018-07-26:
What's your current thinking - Are you going to automate PINN, or duplicate its functionality within each Hypriotos instance?

The latest select and silentreinstallnewer options do almost what you want but only operate on one os, rather than alternating between 2 alternative installations. If you could keep track of which is current and avoid updating both partitions you would be nearly there.

I'm not sure if any of those scripts would be useful hooks for you but your understanding seems ok. I'll answer your questions as best I can off the top of my head.

Background - typ used to show progress.

Pinn_init - typ used to set up display driver.

Before-reboot - executed before EACH reboot into that os ,If it exists. (Merged contribution)

Partition-setup - called once after an os is installed to fix up partition refs - specific to each os.

Noobsconfig - installs extra os customisations per flavour after install but before 1st reboot. Can be used to install scripts that are executed inside an os on 1st or every start of that os or other things.

mortenlj, 2018-07-26:
There are two parts to this in my thinking:

- Initial install
- Atomic upgrades

For the initial install, I think automating PINN would be the way to go, and that's the part I'm mostly focusing on atm.

For the atomic upgrades, it might be that it's easier to duplicate some functionality combined with some smart choices during the initiall install.

procount, 2018-07-26:
I think the initial install can be done with pinn as-is, using a combination of silentinstall, select, pinn.mjh.nz and noobsconfig to install your customisations as a flavour for your atomic upgrades.

When you upgrade, use the autoboot.txt or pinn's sticky default to switch to the new os version's partition.

agherzan, 2018-07-26:
@mortenlj We have in plan to allow custom ssh public keys to get injected in the OS. This will fix the issue you are reporting. On the other hand, the production image allows ssh into the device through VPN.

procount, 2018-07-30:
NOTE: The `before-reboot.sh` script was removed in p2.4.4b because it prevented some OSes from booting.

procount, 2019-03-15:
Any progress on this, or can I close the issue?

mortenlj, 2019-03-15:
I still believe it would be possible, and it would be really cool, but I have found that I don't have time to figure it out. I'll close the issue.

closed 2019-03-15 by mortenlj

procount, 2019-03-15:
OK. If you need any help in the future, just reopen this issue adn I'll see what I can do.