procount/pinn #175: Reinstall and Replace Error - Unable to upgrade to the latest release.

qrao, 2018-07-27:
Reinstall and Replace OS feature has been broken. It appears to be writing to the boot partition, rather than the root partition. I noticed you have made a fix to the reinstall problem in P2.8.5.1. However, I am not able to update to this release on my RPi2. How do I get the hotfix release be detected?

procount, 2018-07-27:
Why can't you upgrade to the new version?

Try going to the maintenance menu. Select PINN as the only checked item then click reinstall. You must be connected to the internet. The new version should be detected and installed.

An alternative is to manually update.
Put your pinn sd card in a PC. Make a copy of recovery.cmdline. dowload pinn-lite.zip and unzip all the contents except recovery.cmdline over the top of your existing pinn installation. If you make a mistake and overwrite recovery.cmdline then restore the copy you made.

Check recovery.cmfline DOES NOT contain 'runinstaller' and delete that option if it exists.

Now put it back in your pi and boot the new version.

procount, 2018-07-30:
You can check the current version number in /mnt/BUILD-DATA

closed 2018-10-04 by procount