procount/pinn #178: SSD USB arch boot fails

lsswizard, 2018-07-31:
Ive installed PINN using SSD without sd card, but when i try to boot Arch2 Os i get this error.

> Not a valid Partition: PARTUUID=55463d29-0a 

but the boot partion is the above, Raspbian boots fine.

procount, 2018-07-31:
You only mentioned 2 OSes - Raspbian and Arch2, but I guess you also have another installed, otherwise that PARTUUID looks a bit wrong to me.
Have you tried installing them on an SD card to see if you get the same behaviour?

lsswizard, 2018-07-31:
yes i have OSMC installed also and that works fine,

Here's my partition list
`/dev/sda1: SEC_TYPE="msdos" LABEL="RECOVERY" UUID="0092-A07D" TYPE="vfat" PARTUUID="55463d29-01"
/dev/sda5: LABEL="SETTINGS" UUID="441fc03b-600f-4968-8d72-3552e12c4181" TYPE="ext4" PARTUUID="55463d29-05"
/dev/sda6: LABEL="boot" UUID="C3C4-BE48" TYPE="vfat" PARTUUID="55463d29-06"
/dev/sda7: LABEL="root" UUID="af4585f8-ff6e-4973-9cdf-e12e84bc7c7f" TYPE="ext4" PARTUUID="55463d29-07"
/dev/sda8: LABEL="boot-rbp2" UUID="33DF-0277" TYPE="vfat" PARTUUID="55463d29-08"
/dev/sda9: LABEL="root-rbp2" UUID="7898d3da-4a25-4d87-9b96-30946a9b3800" TYPE="ext4" PARTUUID="55463d29-09"
/dev/sda10: LABEL="boot_arch" UUID="98A3-A232" TYPE="vfat" PARTUUID="55463d29-0a"
/dev/sda11: LABEL="root_arch" UUID="3101583d-25d1-466b-88c2-c72690afd337" TYPE="ext4" PARTUUID="55463d29-0b"`

and yes it works fine on micro sd.

procount, 2018-07-31:
Well, it sounds like the current Arch2 is no longer compatible with the PARTUUID partition numbering scheme that is used on USB devices. I will test it out when I get a chance to confirm it.

PARTUUID is used for USB devices because it provides a consistent way of identifying the device, rather than the /dev/sdX format that may change on each boot.

If that is the problem, one way round it is to revert the PARTUUID reference to the other format and HOPE that your USB SSD is always /dev/sda. You'll need to edit cmdline.txt in the boot partition and /etc/fstab in the root partition. Replace all occurrences of PARTUUID_55463d29-XX with /dev/sdaYY as given in your partition list

procount, 2018-08-01:
Odd. Those are the only 2 references. I don't think there are any others. Check your edits took effect - you did do a `sync` afterwards, didn't you?

/dev/sd**e**10 - You have a lot of USB devices there!

procount, 2018-08-01:
> `Not a valid Partition: PARTUUID=55463d29-0a`

When/where did that message come up?

lsswizard, 2018-08-01:
Yep did a sync, i always do and just 3 things installed, its how it was setup with PINN.

Just gets to boot screen and i select Arch2 to boot and that messages appears.

lsswizard, 2018-08-01:
OK i used proper partitions and added  /dev/sda11 for cmdline and /dev/sda10 for fstab and same error when i try to boot.