procount/pinn #179: Blank screen

mjmeans, 2018-08-03:
Trying PINN for first time. Using RPi 2, Official RPi 7" Touchscreen, 16 GB SDHC. Here are some issues I encountered.

Reformatted 16 GB SDHC using SDFormater with size adjust, twice to be sure. Copied latest version 2.8.5.1. Booted into PINN. Image on LCD is inverted. Tried editing SDCard /confit.txt from PC added lcd_rotate=2. RPi rainbow screen is now rotated, but PINN is not. Inside PINN went to edit config and added lcd_rotate=2. Now PINN is also rotated. Hurray! Inside PINN edited recovery.cmdline and added vncinstall. PINN no longer boots! The only thing on the screen is:

```
[   2.052309] raspberrypi-firmware soc:firmware: Get Throttled mailbox call fai
led
Starting system message bus:done
```

VNC Viewer is able to connect from PC to RPi, but shows a blank screen and no activity; neither does keyboard work through vnc. Rebooted again and the same result. VNC sees the RPi, but it is a blank screen.

Removed SDCard and edited the recovery.cmdline in a PC using Notepad++(so it would maintain any Unix line endings) to remove vncinstall. Now PINN boots again, but no VNC. So there is some issue with vncinstall and PINN. Or did I do something wrong?

Continuing my testing.... Plugged in Ethernet cable and booted. All looks good right now (except vnc). Selected General tab and Raspbian and Windows 10 IoT Core and changed language to English US/us. Download and install progressed without warnings. After download and install finished, it prompted for image to boot. I selected Raspbian. Raspbian was upside down so I still needed had to edit it's config.txt file to add lcd_rotate=2, but I decided to first use the Raspbian Welcome program to update Raspbian. After Raspbian update finished it prompted to reboot. So I did. I then set the Raspbian /boot/config.txt to add lcd_rotate=2 and rebooted again. All okay.

Next I rebooted to select Windows 10 IoT Core. Windows booting logo started. It was upside down. Sigh. Then I got a blank screen for about 4 minutes before the Welcome to Windows screen appeared displaying "Please wait while we get things ready". It was still upside down. Clearly I have to edit yet another config.txt file for lcd_rotate=2. Unfortunately there is no file editor in Win 10 IoT Core And the Windows IoT Device Portal edits the lcd_rotate=2 setting from the main boot partition. So I had to remove the SDCard and edit the EFIESP partition's config.txt file on regular PC. Finally. All booting now has lcd_rotate=2!

1. Why wasn't vncinstall working in PINN initial boot?

2. I think that if lcd_rotate=x exists in config.txt in the original PINN copied to the SDCard, that it should automatically apply and be added to PINN on first boot. And it should be automatically added to at least Raspbian and Windows 10 IoT OS (and possibly most if not all others) installations' config.txt files if the OS's default config.txt file doesn't already have an entry for lcd_rotate.

procount, 2018-08-04:
1. vncinstall normally works fine. 
Have you read https://github.com/procount/pinn/blob/master/README_PINN.md ? 
There it states that using vncinstall will leave any attached monitor blank, as all display is directed over the VNC connection. Maybe vncshare would be more appropriate for you, as that will simultaneously display the screen on your monitor and over VNC?
You normally need to add forcetrigger in conjunction with the vnc options, otherwise you won't have enough time to press shift to enter PINN before it boots an OS. It normally takes about 6 secs after IP assignment before VNC is active. On first boot, you would have to wait even longer whilst PINN partitions and formats your SD card. The remote_timeout option is also useful for VNC use.
Maybe there is an incompatibility with your VNC viewer? I normally use TightVNC without issue.

2. The upside down nature of the RPF 7" touchscreen is well documented on the forums. e.g. https://www.raspberrypi.org/forums/viewtopic.php?p=821527 . I use the SmartiPi case and mine always displays the right way up in all OSes. Have you considered mounting your LCD screen the other way up? It would avoid having to add `lcd_rotate=2` to every OS (since each is completely independent).
This is a bit of a niche use-case, as it only affects the RPF touchscreen in some display cases, so it is not really a PINN issue. But I agree that processing it automatically in PINN would make it a lot easier for the user. I'm not sure if there are use-cases where this processing shouldn't take place, so it might have to be a configurable option. I shall consider it for a future improvement,

mjmeans, 2018-08-04:
Concerning #1. Yes I had read the readme and was aware of the 6 second issue, but I missed the fact that all display is directed to the VNC and that there is a vncshare option, so I should experiment with that option. However, vncinstall, by its name, suggests that it is for install only, and on first boot, or any time where there is no OS that has been installed, you are clearly "installing". So I have a few suggestions concerning vncinstall:

1. When vncinstall is active and redirecting all UI, maybe it could leave some useful message on the original screen rather than what appears to be an error message: "Get Throttled mailbox failed."
2. When no OS is installed, PINN is clearly in a state that can be considering exclusively for installation. So that is fine there. However, after any OS is installed the PRESS SHIFT screen and the OS boot screens are not "installation" modes per se, but "operation" modes. Perhaps those screens should appear on the normal screen, whether or not vncinstall is present. Only after the user presses SHIFT is the user actually going into a mode where they could perform maintenance and any kind of "install". So perhaps the vncinstall option shouldn't have any effect unless and until PINN reaches that screen. And when it does go to that screen, perhaps is could leave some useful text on the LCD screen rather than an error message.
3. In addition, when PINN detects that an update to itself is, it may still technically be "installing" and OS or OS update, but I think the use case would still benefit from it being on it's main screen instead of being redirected to VNC, so I think #2 suggestion would still apply. i.e. Show the PINN update message and progress on the main screen, not redirected to VNC.
4. Lastly, I do see a use case where a user might want all PINN activity to be redirected to VNC, for example when building an appliance where the end user shouldn't be presented with options to change anything, like for home a media server box. However, in this use case it would be better to leave some useful message on the display, like #1. However, perhaps a better user experience would be for PINN to display a single full page graphic (vncimage.png?) and leave it on the screen while redirecting all other UI to VNC. The end user can then be shown some initial splash screen or boot logo or whatever, until some installed OS begins to load. But still permits VNC controlled maintenance or installation tasks.

procount, 2018-08-04:
Please don't get hung up on the name of 'vncinstall' and that it should only apply to installation modes.
This option is included directly from NOOBS, so you can blame then for the name :wink:

I'm afraid it is an "all or nothing" option and is a feature of the QT framework that the recovery program (i.e. NOOBS or PINN) uses. I just added the `vncshare` variant and `remote_timeout` options to make it a more useful feature.

Although, I agree that a suitable message on the main screen would be helpful to indicate that a VNC connection is required to see the screen in those instances.

procount, 2018-08-08:
VNC message now left on screen when vncinstall is used [p2.8.5.2]

closed 2018-10-04 by procount