procount/pinn #182: Kano OS - no network connectivity

Grottitch, 2018-08-22:
I've tested two 3B devices with PINN and Kano OS, LibreElec and Raspbian installed.

In each case, whilst LibreElec and Raspbian have no issues connecting via either LAN of WiFi, Kano OS refuses to recognise LAN or WiFi.

Strangely, on WiFi, enough information is passing between Kano OS and my router to tell me when I've entered the wrong WiFi password.  But when I enter everything correctly it just times out.

On both my test devices, I have installed Kano OS only (i.e.: directly burned image without PINN) and it connects to LAN or WiFi without issue.  

Can you offer advice as I'm struggling to diagnose the cause of the problem.

Thanks.

procount, 2018-08-22:
It shouldn't be any different.
I'll try it again when I get a chance.
Please confirm you have a 3b and not a 3b plus

Grottitch, 2018-08-22:
Yes, I'm using 3B Pis and not 3B+ Pis.

It is very weird behaviour.  I just tried my third Pi 3B and got the same result.  In this case, Ubuntu Mate booted from PINN had no issue detecting and connecting to the network but Kano OS refuses to connect.  Again, it kicks out an error when the password is input incorrectly, but when it is correct it just times out.

Thanks for your help.  PINN is great!

procount, 2018-08-27:
The version of KanoOS I made available through PINN is beta 3.14.0, whereas the official download has now moved on to beta 3.15.0. Perhaps this is the difference?
I will update to the latest version and see if that fixes it.

matthuisman, 2018-08-28:
Have just updated Kano to 4.0.0.

It's now simply using their NOOBS ready files
https://github.com/matthuisman/pinn-os/blob/master/os_list_v3.json#L239-L270

Tested network on my Pi3B+ and works fine.
So hopefully should also now work on Pi3B.

Grottitch, 2018-09-02:
Many thanks.  It works perfectly now.

One final question if I may: 

When I change the "Destination drive" to a USB drive (as opposed to the SD Card) Kano OS disappears from the list under the Education tab.  

Is this expected behaviour?

procount, 2018-09-02:
Yes. Some OSes do not support installation to USB so that option is disabled.
I'm not sure if v4.0.0 still has a problem with this. Maybe @Matthuisman tested this new version?

closed 2018-10-04 by procount