procount/pinn #186: Can’t get PINN to work with hyperpixel 4.0

Hellindros, 2018-09-10:
I’m having no luck getting PINN to initialise HyperPixel4 correctly.

They even ship files to support the original HyperPixel, and using the same technique results in nada, nothing, zip, zero :(

https://sourceforge.net/projects/pinn/files/

YMMV but it looks like you should be able to extract hyperpixel-display.zip over your PINN recovery partition when you first create the SD card on a host computer and replace the hyperpixel-init binary with the hyperpixel4-init from the HyperPixel4 GitHub - being careful to rename it to hyperpixel-init so it’s called by pinn_init.sh.

Still, it wont work for me and I can’t figure out why. It’s hard to debug with no screen and no UART!

Might be worth raising an issue on the PINN GitHub: https://github.com/procount/pinn 1

lurch, 2018-09-11:
> Might be worth raising an issue on the PINN GitHub

Umm, it looks like you've just done that? :wink:

Hellindros, 2018-09-11:
Yes was tried and also creating a init script with PINN temp name as in the notes , added to the recovery partition , still doesn’t work , any ideas 💡

Hellindros, 2018-09-11:
That sums up the my limited coding skills , hoping you can help , I love the PINN bootloader, and would like to use it to switch between OS via hyperpixel 4.0 instead of the Hdmi that way I’m not tethered to a tv 📺

procount, 2018-09-11:
I've only tested it with the original hyperpixel as that is the only one I have.I imagine the hyperpixel4 will work similarly though.

If you have a network cable and PC you can try debugging it over the network. Add "ssh forcetrigger vncshare' to tecovery.cmdline and ssh into it. Or use vnc to vuew the pinn menu.

procount, 2018-09-12:
I've downloaded the hyperpixel4 github and made all modifications that I think would be necessary to get it to work with PINN. You might like to download it from https://github.com/procount/pinn-os/blob/master/hyperpixel4-display.zip and see if it works for you. Just unzip it over the top of your PINN installation.
I've no way of testing it, so please report back if it works for you.

Hellindros, 2018-09-12:
Great thank you for the code expertise, I will give it a go 👍

Hellindros, 2018-09-13:
Thank you the code works , perfectly, hope it wasn’t too much trouble

closed 2018-09-13 by Hellindros

procount, 2018-09-13:
Yay! First time success. I shall copy it across to Sourceforge next week (as I don't seem to be able to at my current location...)

Hellindros, 2018-09-13:
No problem thank you for making it work , it’s a great bootloader and works well with the hyperpixel 👍