procount/pinn #187: Volumio/Kodi boot related on HDMI TV presence [question]

SCiunczyk, 2018-09-17:
Hi,

I would like to have dual boot system on my RP3 - Kodi/Volumio. But I will be happy to have automatic scenario:

- when HDMI is detected (connected TV is ON) - system boots to Kodi 
- when there is no HDMI detected (TV is OFF) system boots to Volumio (headless)

Do you think that will be possible with PINN ? If yes could you please "push me" in correct direction to build by boot scripts?

best regards
Slawek

procount, 2018-09-17:
That sounds like a novel application, but a bit niche, so I probably won't be including it as standard.

But I think you could achieve it using the `pinn_init.sh` script.
If this script exists on the PINN recovery drive (i.e. the drive where you will find recovery.rfs), it will be executed on boot up before PINN runs.
You can use it to detect the presence of HDMI (see BootSelectionDialog::updateConfig4dsi() for how to use tvservice for this) and alter either the `default_partition_to_boot` or `sticky_boot` parameters in /settings/noobs.conf according to your desires.

SCiunczyk, 2018-09-17:
@procount thanks for your suggestions I will try that. Currently I'm using Volumio with Kodi plug-in but there are lot of problems - eg it's hard to update Kodi in my installation. The system is installed on my boat where I mostly listen to music using Volumio, But time to time I like watch some movies or Netflix so I like to run Kodi with my TV.

I will check and publish my investigation results.

SCiunczyk, 2018-12-27:
Hi, @procount I'm afraid that I need a little more help. I can't get `pinn_init.sh` running on my PINN boot.
My steps are:

1. mount RECOVERY drive: 
`mkdir /tmp/RECOVERY`
`mount -t auto /dev/mmcblk0p1 /tmp/RECOVERY`

2. check if there is recovery.rfs
`cd /tmp/RECOVERY`
`ls recovery.rfs`
exists - OK

3. create pinn_init.sh
`nano pinn_init.sh`

4. enter just two test lines:
`#!/usr/bin/bash`
`echo Hello >/tmp/tv_test.txt`

5. save, exit nano

6. run for test
`./pinn_init.sh`

7. delete test output
`rm /tmp/tv_test.txt`

8. reboot RPi
`reboot now`

and nothing happens. What I'm missing to have `pinn_init.sh` executed on RPi(PINN) boot ?

Happy New Year

procount, 2018-12-27:
So close, yet so far....

In step 4, you need to change the shebang to 
`#!/bin/sh`
because you are in a buildroot environment, not Raspbian. It uses the ash shell, not bash.

It looks like you are trying to do this in Raspbian (or kodi). If you are willing to use VI instead of Nano, you could edit and test this within PINN where you can see what it is doing more easily:

1. Boot into PINN
2. Press Ctrl-Alt-F2 to enter the recovery console
3. Login with root/raspberry
4. The recovery partition is already mounted as /mnt, but it is read-only, so to edit it we need to make it read-write with `mount -o remount,rw /mnt`
5. `cd /mnt`
6. `vi pinn_init.sh`
7. Edit as required and save with `:wq`
8. now you can test it manually using `./pinn_init.sh` and see the output.
9. Don't forget to `sync` to make sure your files are safe before you `rebootp`

SCiunczyk, 2018-12-27:
So close no matter how far... but ... with a little help from my friends ... I have it running now.

Many thanks. I'm very close...

Tomorrow last steps: parsing tvservice output and setting noobs.conf to boot Volumio or Libreelec.

procount, 2018-12-28:
Maybe you can use cec-client to query the status of the tv?
E.g https://blog.gordonturner.com/2016/12/14/using-cec-client-on-a-raspberry-pi/

SCiunczyk, 2018-12-28:
That was my first idea. Is the cec-client available in buildroot environment on boot time ?

Unfortunately my quite modern yacht TV (the target one, with CEC) is currently broken. Have to send it to service.
I have only 20 year old Sony Bravia probably without CEC support.

The whole system is for my boat where I usually listen to music with Volumio and only occasionally I watch movies on Kodi. The TV is closed in cabinet and power off. So this is why I want to have such feature. 99% Volumio and only when cabinet is open and TV is on - Kodi.

Tomorrow I will also check if my ASUS AIO is supporting CEC in HDMI monitor mode.

BTW I think that your function BootSelectionDialog::updateConfig4dsi(QByteArray partition) is also susceptible on that problem. Isn't it?

procount, 2018-12-29:
You would have to add cec-client yourself.

Yes, my bootselection dialog would suffer the same, but it wasn't intended to detect the on/off state, but rather connected or not.

lurch, 2018-12-29:
> The TV is closed in cabinet and power off. So this is why I want to have such feature. 99% Volumio and only when cabinet is open and TV is on - Kodi.

As a workaround hack, instead of detecting whether the TV is on or off, perhaps you could instead detect whether the cabinet door is open or closed, using e.g. a 'switch' made from a couple of bits of tinfoil, connected (by wires) to the RPi's GPIO pins? :slightly_smiling_face:

lurch, 2018-12-30:
...just make sure that the strong magnet doesn't get too close to your HDD! :wink: :skull_and_crossbones:

SCiunczyk, 2018-12-30:
... hi hi hi ... or use SSD 😉

SCiunczyk, 2019-01-02:
Finally I make the switch to cabinet door as @lurch suggest.

Here is my pinn_init.sh:
```
#!/bin/sh
#
# pinn_init.sh
#
# Fayka yHTPC bash script to change default boot image between
# Volumio and LibreELEC
#
# value of RPi GPIO18 pin (input) is used to detect TV cabinet
# door opening state:
#   1 - doors are open - user wants to watch multimedia on LibreELEC
#   0 - doors are open - user wants to listen music on Volumio
#
# depend on pin state script is replacing default noobs.conf file
# with Volumio or LibreELEC version: noobs.conf.vol or noobs.conf.lbe
#
# Script is simple and assume that NOOBS is aviable on /dev/mmcblk0p5
# partition - should be changed to more "inteligent" version.
# For example seeking for noobs.conf file - like PINN itself.
#
# The content of noobs.conf.vol or noobs.conf.lbe could be adjusted
# to particular setup.
# - the important line is default_partition_to_boot=
#   (5 - Volumio, 9 - LibreELEC)
#
# if the user will touch the keyboard or CEC remote - PINN will
# wait for decision as usual.
#
# Wiring - GPIO18 - pin #12 on RPi header - is used as input.
# As detector the simple Reed switch was used. Plus small magnet
# glued to the cabinet door.  Reed switch is pulled-up to 3.3V
# via 47kOhm resitor. There are no possibility to set pull-up/down
# RPi internal resistors via linux shell.
# we try to avoid installing additional libraries on buildroot
# environment. There are no GPIO libraries by default on buildroot
# (RPi.GPIO, WiringPi or similar).
#
# 2019.01.02 by Slawomir Ciunczyk http://www.fayka.pl
#
echo off
dir=$(PWD)

cd /sys/class/gpio

echo 18 > export
echo in > gpio18/direction
cabinetopen=$(cat gpio18/value)

mkdir /tmp/NOOBS
mount -t auto /dev/mmcblk0p5 /tmp/NOOBS

cd /tmp/NOOBS

if [ "$cabinetopen" = "1" ]
then
   echo "TV cabinet open - lets play with LibreELEC"
   rm noobs.conf.bak
   cp noobs.conf noobs.conf.bak
   rm noobs.conf
   cp noobs.conf.lbe noobs.conf

else
   echo "TV cabinet closed - lets play with Volumio"
   rm noobs.conf.bak
   cp noobs.conf noobs.conf.bak
   rm noobs.conf
   cp noobs.conf.vol noobs.conf

fi

cd $dir

sync

umount /tmp/NOOBS
rmdir /tmp/NOOBS
echo on
```
and my hardware (hope that magnet is small and far enough):

[image]
[image]
[image]

Many thanks for Your help.

closed 2019-01-02 by SCiunczyk