procount/pinn #188: Reboot to specified partition from rescue-shell?

Landberg, 2018-09-20:
Is it possible to boot to a different partition from the rescue-shell?  I usually access my RPi through ssh and would like to avoid the GUI for recovery stuff too. 

From bash in an installed OS you can just type `reboot 6` to boot into the 6th partition and bypass PINN. As you can do it from the PINN GUI I thought it would be possible to do it from the rescue-shell too, but I haven't found a simple way to do it yet. I guess it's because the rescue-shell uses the busybox-reboot command and that it doesn't support the extra partition argument.

I've seen some places that it looks possible to set the reboot partition with 

     echo 6 > /sys/module/bcm2708/parameters/reboot_part  

and then call reboot, but that module doesn't exist when I look for it, at least on my RPi 2B v1.1.

Looking at the GUI source-code [.../recovery/util.cpp](https://github.com/procount/pinn/blob/e4dbfd754fb912e0931503894c55438f70316977/recovery/util.cpp#L173), I see it does a syscall specifying the partition. You don't happen to have a command to do that from the rescue-shell too?

The less simple way is to remount partition1, then use your [autoboot.txt bypass file](https://github.com/procount/pinn/blob/e4dbfd754fb912e0931503894c55438f70316977/README_PINN.md#how-to-bypass-pinn-and-boot-directly-into-a-fixed-partition) feature. It also needs to be removed once I'm in the OS I selected so I don't permanently bypass PINN. 

Do you have a simpler solution for my problem?

Keep up the good work!

procount, 2018-09-20:
Maybe this will work?
https://www.raspberrypi.org/forums/viewtopic.php?t=71107#p1321117
You might have to remove the system call for init.d.

Landberg, 2018-09-21:
Thanks, as you said I had to modify it a little to compile on my Arch Linux, but it works fine.

This is the code I ended up with in main.cpp:
```
#include <linux/reboot.h>
#include <sys/syscall.h>
#include <unistd.h>

int main(int argc, char *argv[])  {
    char *param = NULL;
    int cmd;

    if(argc == 2) {
        param = argv[1];
        cmd = LINUX_REBOOT_CMD_RESTART2;
    }
    else {
        cmd = LINUX_REBOOT_CMD_RESTART;
    }

    sync();
    syscall(SYS_reboot, LINUX_REBOOT_MAGIC1, LINUX_REBOOT_MAGIC2, cmd, param);

    return 0;
}
```

Compile it with:
```
gcc -static main.cpp -o part_reboot
````

I then put **part_reboot** in the root folder of the PINN file-tree. I.e.: together with recovery.cmdline, recovery.img, etc.

Now I can call `/mnt/part_reboot 6` from the rescue-shell to boot into my partition.

To get a clean exit from the ssh session, to avoid having to wait for the timeout,  you can call it like this instead:
```
(sleep 1 && /mnt/part_reboot 6 &) && exit
```

procount, 2018-09-21:
Cool! Glad it worked.
I shall have to compile it into PINN's buildroot.

Landberg, 2018-09-21:
That would be great!

procount, 2018-10-04:
Included in p2.8.5.5 with slightly different rebootp.sh, but I still don't think it works as expected. Maybe it can't, but released anyway, so closing.

closed 2018-10-04 by procount

lurch, 2018-10-04:
I just did a quick search myself, and I wonder if the SSH disconnect problem might be fixed by explicitly killing the sshd daemon before calling the reboot command? (i.e. perhaps calling reboot directly doesn't allow sshd to shut down cleanly and close remote connections?)

procount, 2018-10-04:
Hey @Lurch - good to see you're still keeping a watchful eye over me!
Yeah, sounds like a reasonable approach. I'll have to give it a try.

procount, 2018-10-04:
How about....

```
#!/bin/sh
killall dropbear && /usr/bin/rebootp $1 &
```

lurch, 2018-10-04:
Haha, only very occasionally :wink: 

I don't have time to test myself, but does `killall` actually wait for the process to finish, or does it just send the KILL signal? You might need to put a `sleep` in there to give `dropbear` time to shut down gracefully?

procount, 2018-10-04:
`killall` works just like `kill` so it sends a SIGTERM signal by default, but uses the process name as a parameter instead of a PID. 
It seems to work without the `sleep`, since it causes an abrupt termination and forces the client to drop the connection immediately, which it does virtually instantly, so I think it is sufficient. But yeah, it could be changed to the following for a small delay:
```
killall dropbear && (sleep 1; /usr/bin/rebootp &) &
```

maovidal, 2022-10-15:
Hi everyone:

Both the `rebootp` at: https://github.com/procount/pinn/tree/master/rebootp and the `Buildroot package` at: https://github.com/procount/pinn/tree/master/buildroot/package/rebootp are helpful to provide a reboot mechanism for cases like the one described here: https://forums.raspberrypi.com/viewtopic.php?t=341372 where the `reboot` command does not work in the default `Buildroot` board definitions for Raspberry Pi.

I would like to propose you to make a standalone `rebootp` repository so others like me can include it as a custom `buildroot` while keeping track of its origins. 

If that is appealing to you and you decide to create the repo in one your accounts (if preferred), I could make the pull requests on this repo to include that external new repo.