procount/pinn #189: Sticky Default OS change

Haris1977, 2018-09-24:
I really thing the "Sticky Default OS" should be somehow altered.

I know how it works now, but i do not like the idea of bypassing the 10 sec bootmenutimeout option. I think it is better to have the possibility (if we want) to use both options together (like berryboot does). Sticky default OS + the 10 second boot time menu option. In that way PINN will always boot to the OS that we prefer AND give us the 10 sec time to alter to a different OS when we want. And - the most crucial - ...in next reboot sticky default OS will be booted again !

P.S1 E.g 95% of my time i want to boot into LibreELEC. But, if i change that and boot into let's say Raspbian, PINN will remember this last OS choice and boot into Raspbian by default each time from now on, unless a different option is selected. So i need to boot again into LE to correct this..This is just wrong :)

PS2 I know i can use the shift button to temporarily boot into another OS but..it is a little cumbersome for some ppl like me (i do not use a keyboard, only my remote)

So @procount can you add that option for the ppl who want both sticky default and bootmenutimeout?

PS. I have downloaded the 2854a.zip (rpi3) but it doesnt work.

I ve installed LE+Raspbian (as a test), added the bootmenutimeout=5 in cmdline and ticked LE as "sticky default".

When i started raspbian - rebooted it - let it see what happens after 5 second and..raspbian again :arrow: (it seems as sticky deafault is overrided by the boot time menu option). I do see though that LE is blue highlighted

procount, 2018-09-24:
working on it...

procount, 2018-09-24:
Yes, I confirm the last booted OS seems to be getting priority. 
BTW, is the CEC problem fixed, as you didn't copy the focus problem here from your post?

Haris1977, 2018-09-24:
You can have a different approach if y want here: We know that by holding the "shift" key PINN's menu is activated, right? Nice..My though then: You could implement an other key - lets say "Key_M" or an other doesnt matter ("M" for menu maybe? - i see the key codes from cec_keys.md) **### to "activate" OS boot menu** .. So if this can be happen we can then key map "Key_M" to a corresponding remote control button (e.g blue, red, green, yellow button) in order to bring up the OS Boot menu selector only when we want!!

procount, 2018-09-24:
You already have that.
PINN will recognise any CEC key in place of the shift key. You can also use the left mouse button to click on the splash screen to enter the recovery memu.

Haris1977, 2018-09-25:
I dont talk about the recovery menu. I am talking about the OS boot menu...

Haris1977, 2018-09-25:
By digging a little more i ve managed to enter the OS Boot menu only with my remote. First i removed the bootmenutimeout=5 from the cmdline and made my LibreELEC image as "sticky default". Then i reached the OS boot menu by:

1) Pressing any key on my remote (i use the red button) during the boot up process
2) Waiting 8-10 sec for the pinn to initialize (unluckily PINN needs some time cause it searches the internet for the files)
3) Pressing again the red button two times to cycle through tabs until i see the Esc tab. (Thankfully next time pinn will remember it and you can ommit step 3)
4) Pressing the 5 key on my remote

Then i have the OS boot menu :)

So, i use a "sequence of keys" to enter the OS boot menu only with my remote control. 

I could avoid all that by just hitting the M key on my keyboard (or red button for my remote) during boot up. That is my thought:)

procount, 2018-09-25:
Have you mapped one of your CEC keys to the 'Esc' key? Let's assume this is your red button. If so I expected the following key sequence to be used:
1. When the wallpaper appears ("Press shift" screen), press the red button to enter the recovery menu.
2. When the "Wait for pinn to initialise" dialog box appears as it connects to the internet, just press the red key again to cancel this dialog.(You don't have to wait for it to complete)
3. Now just press the red key a third time to exit to the boot menu screen.
4. Select your OS with up & down and press the button mapped to the Enter key.

So steps 1-3 are just 3 quick presses of the red button to get to the boot selection menu. Not as quick as having a boot menu timeout, but if 99% of the time you are using the same OS, the normal case is quicker, and selecting another OS for the other 1% of the time is not too difficult.

Haris1977, 2018-09-25:
Inside my cec_json file and in "any" tab, my red button is mapped to key "M" (default). Key "Escape" is the "CEC_User_Control_Exit" (which i do not use it as i only use up/down/select). These are my 2 (among other) key settings (in the "any" tab i repeat):

"Key_Escape":   "CEC_User_Control_Exit",
"Key_M":        "CEC_User_Control_F2Red"

But in my "main menu" tab i also have

"Key_Escape":   "CEC_User_Control_Number5"

Which one of the 2 tabs should i change? Any tab or Main menu tab?

Haris1977, 2018-09-25:
I dont have an exit on the specific tv. Only back. Is back key code supported?

procount, 2018-09-25:
I've no idea what CEC codes your remote provides. It doesn't really matter which key you use, as long as you map it to the Escape Key. You'll need to experiment. 

If you can use a full keyboard with PINN you can use the recovery shell (Press ctrl-alt-F2 and login with root/raspberry), or you can ssh into it from another PC if it is on the network(same username/password).
Then type 
```
tail -f /tmp/debug
```
This will show the PINN debug messages. 
Each time you press a recognised key on your CEC remote, it will come up with a message like:
```
CEC key: 10 CEC_User_Control_Exit
```
So you can use this information to determine which keys are usable and what their keycodes are so you can map them to PINN keyboard keys in your own cec file. Just choose a recognised key that makes sense to you for the Exit function.

Haris1977, 2018-09-25:
Thanks for the info. I was playing with the hdmi cec options and now my tv's hdmi cec doesnt work anymore:( Oh God.. I thing smth conflicted my tv here :(

Haris1977, 2018-09-25:
Of course yes ;) But i had to totally unplug the power from my tv for some minutes to make it work

procount, 2018-09-25:
Tricky things, these CEC devices!

Haris1977, 2018-09-25:
Indeed. I am closing the issue then. Thanks a lot again!

procount, 2018-09-25:
So you got it working with a key mapping to the ESC key?
I may continue this work anyway. I think I just missed a pathway through the code.

Haris1977, 2018-09-25:
Kind of. I mapped my Red remote button ("CEC_User_Control_F2Red") to **every** "Key_Escape" key code. I did this for all of the tabs (main menu, archival, maintenance etc). But there is a conflict now if i always use my red button. So - i overcame it - by just hitting an other button on my remote when shift key needed to be pressed (the select button) and then 2 times the red one (one to stop initializing and an other one to go to the OS boot menu). So for the time - i think - i am ok. 

My json file is here if y want to take a look: https://www25.zippyshare.com/v/4bTQk1zB/file.html

Haris1977, 2018-09-25:
I will stop fiddling with the cec_json file cause i had really bad crashes of my TV's hdmi cec. I am staying with v2.8.5.4 and my original json file until a better solution comes.

I m closing the topic for now, thanks :)

procount, 2018-09-26:
Try /testing/pinn-2854b.zip

Haris1977, 2018-09-26:
Yes! This is working as expected , thanks :)

The only thing that still remains (at least in my thoughts) is the OS renaming option :)

procount, 2018-09-26:
Great. I'll close this now.
Patience is a virtue.... :wink:

closed 2018-09-26 by procount