procount/pinn #190: Please update Gentoo64 images (v1.3.0 released)

sakaki-, 2018-09-26:
Hi @procount -

I've just released **v1.3.0** of the [gentoo-on-rpi3-64bit](https://github.com/sakaki-/gentoo-on-rpi3-64bit) image (including Pi-Top variant), and redone the first slide of the marketing deck to match (available [here](https://github.com/sakaki-/resources/blob/master/raspberrypi/pi3/noobs/Slide1.png)).

Whenever you get a chance, would you mind updating the images on PINN?

Many thanks for your help!

sakaki

procount, 2018-09-27:
Sure thing.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/procount/pinn/issues/190",
"url": "https://github.com/procount/pinn/issues/190",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Please update Gentoo64 images (v1.3.0 released) (#190)",
"sections": [
{
"text": "",
"activityTitle": "**sakaki**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@sakaki-",
"facts": [
{
"name": "Repository: ",
"value": "procount/pinn"
},
{
"name": "Issue #: ",
"value": 190
}
]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"procount/pinn\",\n\"issueId\": 190,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"procount/pinn\",\n\"issueId\": 190\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/procount/pinn/issues/190"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 384664221\n}"
}
],
"themeColor": "26292E"
}
]

procount, 2018-09-30:
Done

closed 2018-09-30 by procount

sakaki-, 2018-09-30:
Thanks! I've [added a short post noting this](https://www.raspberrypi.org/forums/viewtopic.php?p=1371709#p1371709) to the announce thread.