procount/pinn #191: Re-install PINN without formatting entire SD-card

Kohlroulade, 2018-10-03:
I recently crashed my PINN without exactly knowing how (see https://raspberrypi.stackexchange.com/questions/89319/install-pinn-without-deleting-oses-on-sd-card for details). I get only rainbow-screen together with the small thunder-symbol at the upper right corner. However the power-supply is still the same as before when it worked (officially supplied RPi-power-supply with 5.1V), so I doubt that issue is in any way related to to less current.

So my question is if it is possible to re-install PINN without deeting all the data on my two different OSes (Raspian and RetroPie)?

procount, 2018-10-03:
> I have a RPi 3B+ on which I installed two OSes via PINN. That worked fine until now.
> Recently I tried to configure my bootloader so I am able to chose the correct OS to boot from my TVs remote-control. 

Do you mean using CEC controls?

> As I didn´t find any option on the PINN-user-interface to do so 

There are details in [README_PINN.md](https://github.com/procount/pinn/blob/master/README_PINN.md#cec-control) and some alternative key mappings on my [wiki](https://github.com/procount/pinn/wiki/CEC-Mappings-where-cursor-keys-navigate-the-tabbed-lists). You have to manually edit the key mappings, I'm afraid.

>I pressed Alt+Ctrl+F4 which should have rebooted the Pi.

Ctrl-Alt-F4 does NOT reboot a Pi (In Windows it will close a window). On a RPi it switches to an alternative login terminal. In PINN, I think there are only 4 enabled: ctrl-Alt-F1 (Pinn's GUI) to ctrl-alt-f4.

> However now I get only the rainbow-screen together with the power-symbol in the upper right corner. As I´m using the exact same hardware as all the time before, I can´t imagine the power-supply is low. In particular as I use the official power-supply with 5.1V output. Instead I assume I crashed the PINN in some way.

Not sure what you did, but it shouldn't be due to ctrl-alt-F4. You probably have the wrong firmware loaded.
Is the red or green LED on the RPi flashing in a particular way, like 4 quick flashes then 4 slow flashes?
Did you accidentally select the PINN menu option to downgrade the firmware?
Have you recently upgraded from an older RPi to the 3B+?
Do you have an older RPi that you can put the PINN SD card in and see if it boots on that?

> So I wondered if I can re-install PINN on my SD-card without needing to re-install the OSes. 

Probably Yes, but please answer the above questions first.

> However when I put the card into my Windows-PC I see several drives named boot_rp44 (probably the RetroPies boot-section) and boot (probably the raspians boot-section) together with one drive named RECOVERY and multiple ones that have only a drive-letter which I can´t access at all (wndows asks me to format those drives).

That is as expected. Windows can only read the FAT partitions, but not the other EXT4 formatted partitions that Linux uses (unless you install specific drivers for them). DO NOT let Windows format them otherwise you WILL lose your OSes!

procount, 2018-10-03:
OK. It looks like you have accidentally downgraded the PINN firmware. It is fixable. I will write some instructions for you.

The RECOVERY partition is the first partition on the SD card and is where PINN is stored. You will need to be able to see its contents from Windows and replace some files to fix your problem. You should see many files, including e.g. recovery.rfs.

Kohlroulade, 2018-10-03:
Hmm, strange but anyway.

Yes, I see the recovery.rfs together with recovery.elf and .img.

Kohlroulade, 2018-10-03:
> Copy these 2 files to the root directory

You mean the (in my case two) root-drives "root" (=raspian) and "rootrp44" (=RetroPie) or which directory you refer to?

procount, 2018-10-03:
No, no, no.  This is all only done in the RECOVERY drive partition where PINN is stored (which is why I said "where recovery.rfs is")
You should leave the partitions where your OSes are stored as they are.

Kohlroulade, 2018-10-03:
You are my hero. Worked perfectly. Thank so much.

closed 2018-10-03 by Kohlroulade

procount, 2018-10-03:
You're welcome!

There is a menu button on the Archival menu labelled "Firmware(f)". 
In the case of your 3B+, this should always show a green arrow pointing down with "3B+" written across it in red. Please make sure it always stays in this state. Do not click it. If you do accidentally click it, click it again to revert it before you exit PINN. If you are using a TV remote, it can also be clicked by pressing key number '4' on the TV remote whilst in this menu. If it does happen again, just repeat the process above.

procount, 2018-10-04:
In the next version of PINN, the firmware button will be disabled on the 3B+ because it serves no purpose

lurch, 2018-10-04:
...and presumably at some point in the future all of the OSes used by PINN will be "new enough" that the "firmware button" will become unnecessary and can be removed? :slightly_smiling_face: