procount/pinn #194: (NOT BUG) Update for Lineage

BlazingOne, 2018-10-31:
Hey there, this is a request for https://konstakang.com/devices/rpi3/CM14.1/ (lingeage) to be updated to it's latest version made in august, hope you can do that :)

procount, 2018-10-31:
I started to convert this image, but I had a problem with sourceforge falsely detecting something as malware. This took a while to sort out, but now it is ok. Sorry, I've been busy developing PINN for the next version and haven't had time to get back to releasing these.

If you want to try it out, you can find a copy in https://sourceforge.net/projects/pinn/files/os_next/ToCheck/lineage/
If you do, please provide some feedback whether it is ok.

procount, 2020-04-02:
Updated to v16

closed 2020-04-02 by procount