procount/pinn #195: How to restore dead SD card used to boot a HDD?

procount, 2018-11-03:
Pro count uh I need help. So basically last night I shutdown my raspberry pi 3B+, turned it on this morning and nothing happens. I plugged the SD card into a reader on my Mac and the card could not be initialised. I am thinking the card is dead. Basically I had the boot on the SD card, and installed the OS on a HDD. The HDD still works. How can I get the Raspberry pi working again?

Thanks
DarkPlatinim

darkplatinum, 2018-11-03:
Ok I'll get a 64 GB sad card from tesco, will a SanDisk ultra be ok? It's £17 compared to the extreme edition which is £28

procount, 2018-11-03:
Sandisk ultra is ok.
Sizewise is up to you.
What size was your bad one?
If you are only going to use it to boot your hard disk then 64gb would be a waste, ultimately. But you do need something big enough temporarily to store the 2 raspbians and retorpie onto (not incl. Game roms). 16gb might be ok. 32 would definitely be big enough. 
If you bought a 64gb card, I would use it temporarily to install the oses and create your boot disk. Then copy the boot disk to a smaller 8gb card to boot the hard drive and use the 64gb card for something else.

If all 3 oses can install to a 16gb card, then I might just use one of those and then keep it as a boot card aswell.

Depends on how big your pockets are and what you're going to use them for afterwards.

Sorry, but I don't have time to test how big a card you need at the moment (busy weekend) but I'll write some instructions as soon as possible.
Step 1 is get sd card.
Step 2 install Raspbian on sd card so you can access your hard disk and get some info off it.
Step3 remove hdd
Step4 install pinn on new sd card
Step 5 install 2x raspbian and retropie as before but onto sd card
Step 6 edit sd card to boot OSes off hdd.
Step 7 (optional) remove rootfs from sd card and just leave boot partitions. Then you can copy it to a smaller boot only sd card.

Let me know when you have done step 2

darkplatinum, 2018-11-03:
My previous one was 8 GB. I'll get a 32 GB SD card. Do I have to start a fresh installation of Raspbian? I can't just use the OS on the HDD?

procount, 2018-11-03:
It has no boot partition.
Actually you could just install pinn to 32gb card. Then install 2 raspbian & retropie to 32gb card. Then you can use one of those raspbians to probe your hdd. That would save a step.

darkplatinum, 2018-11-03:
Hmm... what if I used your tutorial https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN

?

darkplatinum, 2018-11-03:
So basically what you mean is 1. Download PINN on SD card, 2. Install 2 raspbian and 1 retro pie to SD card from online... Then?  Also my HDD has multiple OS on it, i just use 2 raspbian and retropie mainly. The others are extra. I have several project spaces as well.

darkplatinum, 2018-11-03:
I have downloaded: 2x Raspbian, 1x RetroPi, 1x LibreELEC And 7x project spaces on SD card.

procount, 2018-11-03:
What you need to do is to install on the SD card EXACTLY what you installed on your HDD. 
More OSes would not be a problem, but you need to at least install what you had on the HDD.

When you've done that we will examine your HDD to find out which partitions have which OSes installed on them, then we will rebuild the data files for PINN so that it nows about them.

darkplatinum, 2018-11-03:
The thing is I've forgotten EXACTLY what OS I have downloaded on my SD card. Looking at the filesystem on the HDD I can see what exactly is retropie, and the raspbians. Would it be possible to delete some of the filesystems that I don't really need and add it to the free space that is automatically put in the filesystem?

procount, 2018-11-03:
No matter. Will find out and adjust if necessary.
Have you now installed PINN and at least Raspbian?
When you have done that, boot PINN and select one of the Raspbians and boot into that.
In Raspbian open a terminal.
Enter the following commands and cut&paste the output of each command here.

```mount```
```sudo fdisk -l```

darkplatinum, 2018-11-03:
Here is `mount` WITHOUT HDD plugged in: 
`pi@raspberrypi:~ $ mount
/dev/mmcblk0p9 on / type ext4 (rw,noatime,data=ordered)
devtmpfs on /dev type devtmpfs (rw,relatime,size=470116k,nr_inodes=117529,mode=755)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/net_cls type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=34,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
mqueue on /dev/mqueue type mqueue (rw,relatime)
sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
/dev/mmcblk0p8 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=94944k,mode=700,uid=1000,gid=1000)
fusectl on /sys/fs/fuse/connections type fusectl (rw,relatime)
gvfsd-fuse on /run/user/1000/gvfs type fuse.gvfsd-fuse (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
/dev/mmcblk0p7 on /media/pi/root_rp44 type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/mmcblk0p6 on /media/pi/boot_rp44 type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
/dev/mmcblk0p5 on /media/pi/SETTINGS type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/mmcblk0p21 on /media/pi/prjroot3 type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/mmcblk0p20 on /media/pi/prjboot3 type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
/dev/mmcblk0p19 on /media/pi/prjroot2 type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/mmcblk0p18 on /media/pi/prjboot2 type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
/dev/mmcblk0p17 on /media/pi/prjroot1 type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/mmcblk0p16 on /media/pi/prjboot1 type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
/dev/mmcblk0p15 on /media/pi/prjroot0 type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/mmcblk0p14 on /media/pi/prjboot0 type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
/dev/mmcblk0p13 on /media/pi/prjroot type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/mmcblk0p12 on /media/pi/prjboot type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
/dev/mmcblk0p11 on /media/pi/Storage type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/mmcblk0p10 on /media/pi/System type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)`

fdisk -l:
`pi@raspberrypi:~ $ sudo fdisk -l
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

[... 59 lines cut ...]
Disklabel type: dos
Disk identifier: 0x07189166

[... cut ...]
/dev/mmcblk0p14      36495360 36700157   204798  100M  c W95 FAT32 (LBA)
/dev/mmcblk0p15      36700160 42942461  6242302    3G 83 Linux
/dev/mmcblk0p16      42942464 43147261   204798  100M  c W95 FAT32 (LBA)
/dev/mmcblk0p17      43147264 49389565  6242302    3G 83 Linux
/dev/mmcblk0p18      49389568 49594365   204798  100M  c W95 FAT32 (LBA)
/dev/mmcblk0p19      49594368 55836669  6242302    3G 83 Linux
/dev/mmcblk0p20      55836672 56041469   204798  100M  c W95 FAT32 (LBA)
/dev/mmcblk0p21      56041472 62333951  6292480    3G 83 Linux
pi@raspberrypi:~ $ 
`

procount, 2018-11-03:
Sorry. I should have said to plug your hdd in because that is what I need to look at.

procount, 2018-11-03:
I may be a bit slow in responding as I'm busy, but don't worry. We'll get there.

darkplatinum, 2018-11-03:
Not a problem.

mount: 
`pi@raspberrypi:~ $ mount
/dev/mmcblk0p9 on / type ext4 (rw,noatime,data=ordered)
devtmpfs on /dev type devtmpfs (rw,relatime,size=470116k,nr_inodes=117529,mode=755)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/net_cls type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=34,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
mqueue on /dev/mqueue type mqueue (rw,relatime)
sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
/dev/mmcblk0p8 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=94944k,mode=700,uid=1000,gid=1000)
fusectl on /sys/fs/fuse/connections type fusectl (rw,relatime)
gvfsd-fuse on /run/user/1000/gvfs type fuse.gvfsd-fuse (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
/dev/mmcblk0p7 on /media/pi/root_rp44 type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/mmcblk0p6 on /media/pi/boot_rp44 type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
/dev/mmcblk0p5 on /media/pi/SETTINGS type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/mmcblk0p21 on /media/pi/prjroot3 type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/mmcblk0p20 on /media/pi/prjboot3 type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
/dev/mmcblk0p19 on /media/pi/prjroot2 type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/mmcblk0p18 on /media/pi/prjboot2 type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
/dev/mmcblk0p17 on /media/pi/prjroot1 type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/mmcblk0p16 on /media/pi/prjboot1 type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
/dev/mmcblk0p15 on /media/pi/prjroot0 type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/mmcblk0p14 on /media/pi/prjboot0 type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
/dev/mmcblk0p13 on /media/pi/prjroot type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/mmcblk0p12 on /media/pi/prjboot type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
/dev/mmcblk0p11 on /media/pi/Storage type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/mmcblk0p10 on /media/pi/System type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
/dev/sdb10 on /media/pi/Storage1 type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/sdb6 on /media/pi/root type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/sdb9 on /media/pi/root_raspx type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/sdb11 on /media/pi/root-rbp2 type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/sdb8 on /media/pi/root0 type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/sdb13 on /media/pi/prjroot01 type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/sdb12 on /media/pi/prjroot4 type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/sdb18 on /media/pi/prjroot5 type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/sdb14 on /media/pi/prjroot11 type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/sdb16 on /media/pi/prjroot31 type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
/dev/sdb17 on /media/pi/prjroot41 type ext4 (rw,nosuid,nodev,relatime,data=ordered,uhelper=udisks2)
[... cut ...]
/dev/sdb12       869244928 1004314621  135069694  64.4G 83 Linux
/dev/sdb13      1004314624 1139384317  135069694  64.4G 83 Linux
/dev/sdb14      1139384320 1274454013  135069694  64.4G 83 Linux
/dev/sdb15      1274454016 1409523709  135069694  64.4G 83 Linux
/dev/sdb16      1409523712 1544593405  135069694  64.4G 83 Linux
/dev/sdb17      1544593408 1679663101  135069694  64.4G 83 Linux
/dev/sdb18      1679663104 1814732797  135069694  64.4G 83 Linux
/dev/sdb19      1814732800 1953525166  138792367  66.2G 83 Linux
pi@raspberrypi:~ $ 
`

procount, 2018-11-03:
Can you post the contents of /media/pi/SETTINGS/installed_os.json plesae?

darkplatinum, 2018-11-03:
[
 {
  "bootable" : true,
  "description" : "Retropie2 v4.4 gaming distro using Emulation Station for RPi2/3",
  "folder" : "/settings/os/Retropie2",
  "group" : "Games",
  "icon" : "/settings/os/Retropie2/icon.png",
  "name" : "Retropie2",
  "partitions" : [
"/dev/mmcblk0p6",
"/dev/mmcblk0p7"
  ],
  "password" : "raspberry",
  "release_date" : "2018-04-14",
  "supported_models" : [
"Pi 2",
"Pi 3"
  ],
  "url" : "https://retropie.org.uk/",
  "username" : "pi"
 },
 {
  "bootable" : true,
  "description" : "A port of Debian Stretch for the Raspberry Pi (full desktop version)",
  "folder" : "/settings/os/Raspbian",
  "group" : "",
  "icon" : "/settings/os/Raspbian/icon.png",
  "name" : "Raspbian",
  "partitions" : [
"/dev/mmcblk0p8",
"/dev/mmcblk0p9"
  ],
  "password" : "raspberry",
  "release_date" : "2018-10-09",
  "supported_models" : [
"Pi Model",
"Pi 2",
"Pi Zero",
"Pi 3",
"Pi Compute Module 3"
  ],
  "url" : "http://www.raspbian.org/",
  "username" : "pi"
 },
 {
  "bootable" : true,
  "description" : "LibreELEC is a fast and user-friendly Kodi Entertainment Center distribution.",
  "folder" : "/settings/os/LibreELEC_RPi2",
  "group" : "",
  "icon" : "/settings/os/LibreELEC_RPi2/icon.png",
  "name" : "LibreELEC_RPi2",
  "partitions" : [
"/dev/mmcblk0p10",
"/dev/mmcblk0p11"
  ],
  "password" : "libreelec",
  "release_date" : "2018-04-13",
  "supported_models" : [
"Pi 2",
"Pi 3"
  ],
  "url" : "",
  "username" : "root"
 },
 {
  "bootable" : false,
  "description" : "A Project Space to partition the SD card equally",
  "folder" : "/mnt/os/ProjectSpace",
  "group" : "Utility",
  "icon" : "/mnt/os/ProjectSpace/ProjectSpace_-_1.png",
  "name" : "ProjectSpace - 1",
  "partitions" : [
"/dev/mmcblk0p12",
"/dev/mmcblk0p13"
  ],
  "password" : "?",
  "release_date" : "2018-05-14",
  "supported_models" : [
"Pi Zero",
"Pi Model",
"Pi Compute Module",
"Pi 2",
"Pi 3"
  ],
  "url" : "https://",
  "username" : "?"
 },
 {
  "bootable" : false,
  "description" : "A Project Space to partition the SD card equally",
  "folder" : "/mnt/os/ProjectSpace",
  "group" : "Utility",
  "icon" : "/mnt/os/ProjectSpace/ProjectSpace_-_2.png",
  "name" : "ProjectSpace - 2",
  "partitions" : [
"/dev/mmcblk0p14",
"/dev/mmcblk0p15"
  ],
  "password" : "?",
  "release_date" : "2018-05-14",
  "supported_models" : [
"Pi Zero",
"Pi Model",
"Pi Compute Module",
"Pi 2",
"Pi 3"
  ],
  "url" : "https://",
  "username" : "?"
 },
 {
  "bootable" : false,
  "description" : "A Project Space to partition the SD card equally",
  "folder" : "/mnt/os/ProjectSpace",
  "group" : "Utility",
  "icon" : "/mnt/os/ProjectSpace/ProjectSpace_-_3.png",
  "name" : "ProjectSpace - 3",
  "partitions" : [
"/dev/mmcblk0p16",
"/dev/mmcblk0p17"
  ],
  "password" : "?",
  "release_date" : "2018-05-14",
  "supported_models" : [
"Pi Zero",
"Pi Model",
"Pi Compute Module",
"Pi 2",
"Pi 3"
  ],
  "url" : "https://",
  "username" : "?"
 },
 {
  "bootable" : false,
  "description" : "A Project Space to partition the SD card equally",
  "folder" : "/mnt/os/ProjectSpace",
  "group" : "Utility",
  "icon" : "/mnt/os/ProjectSpace/ProjectSpace_-_4.png",
  "name" : "ProjectSpace - 4",
  "partitions" : [
"/dev/mmcblk0p18",
"/dev/mmcblk0p19"
  ],
  "password" : "?",
  "release_date" : "2018-05-14",
  "supported_models" : [
"Pi Zero",
"Pi Model",
"Pi Compute Module",
"Pi 2",
"Pi 3"
  ],
  "url" : "https://",
  "username" : "?"
 },
 {
  "bootable" : false,
  "description" : "A Project Space to partition the SD card equally",
  "folder" : "/mnt/os/ProjectSpace",
  "group" : "Utility",
  "icon" : "/mnt/os/ProjectSpace/ProjectSpace_-_5.png",
  "name" : "ProjectSpace - 5",
  "partitions" : [
"/dev/mmcblk0p20",
"/dev/mmcblk0p21"
  ],
  "password" : "?",
  "release_date" : "2018-05-14",
  "supported_models" : [
"Pi Zero",
"Pi Model",
"Pi Compute Module",
"Pi 2",
"Pi 3"
  ],
  "url" : "https://",
  "username" : "?"
 }
]

procount, 2018-11-03:
My first guess is that you had the following installed on your HDD:
Raspbian
Gentoo64
Raspbian
Raspex
OpenELEC or LibreELEC?
rbp2?
8 project spaces

I can't remember what rb2 is - I'll have to look it up, but does that sound right? PLesae correct any Ihave got wrong.

darkplatinum, 2018-11-03:
2 x raspbian yes, Gentoo pretty sure raspex think so... LibreELEC yes, rbp2 pretty sure remebering downloading another OS but not sure what it was. I know it was one of the desktop versions though, not CLI.

procount, 2018-11-03:
We'll need to go through them in turn to make sure what they are.
Can you post the contents of /media/pi/root/etc/fstab?

darkplatinum, 2018-11-03:
proc            /proc           proc    defaults          0       0
PARTUUID=c27920ee-06  /boot           vfat    defaults          0       2
PARTUUID=2b2efe32-06  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that

procount, 2018-11-03:
I'm going to keep looking to see what rbp2 is, but meanwhile, maybe you can have a browse in /media/pi/root-rbp2 and see if you can identify what it is.

darkplatinum, 2018-11-03:
Sure.

darkplatinum, 2018-11-03:
/media/pi/root is retropie, /media/pi/root0 is Raspbian #1, /media/pi/root-rbp2 is Raspbian #2.

darkplatinum, 2018-11-03:
Going to turn in for the night. Will be on tommorow.

procount, 2018-11-03:
Really? you sure? That's not what I expected. 
But Raspbian might be renaming the mountpoints to avoid having 2 mountpoints called the same name.

How did you get 2 versions of Raspbian on there anyway? Is one full Raspbian and the other Raspbian-Lite?

procount, 2018-11-03:
OK. tomorrow.

darkplatinum, 2018-11-03:
Yes I am sure, the second raspbian had some windows 98 QEMU files. Found them in there.They are both full Raspbians as well. Maybe something to do with having 2 of exactly the same images? 

Goodnight

procount, 2018-11-04:
So I this is what you need to do.
Remove HDD (to avoid making mistakes)
Boot PINN on your SD card and install the following OSes on to it (to match what is on your HDD and removing the previous ones you put on the SD card)

Retropie
Gentoo64
Raspbian
Raspex
LibreELEC
PiTop (just temporarily)
ProjectSpace1
ProjectSpace2
ProjectSpace3
ProjectSpace4
ProjectSpace5
ProjectSpace6
ProjectSpace7
ProjectSpace8

Once installed, replace PiTop with another copy of Raspbian (so you have 2 Raspbians)

Then boot into one of the Raspbians and paste a new copy of /media/pi/SETTINGS/installed_os.json here.

procount, 2018-11-04:
The next step is to match up the existing partitions on your HDD with the new OSes you've just installed on your SD card.

Your previous posts had the HDD mounted as /dev/sdb, but I didn't see a /dev/sda - maybe that was just the way it happened.

I will assume though that if you start PINN with the HDD as the only USB device, that it will be mounted as /dev/sda. So I used that notation below.

I have replaced the device for the root partition for each OS with what I THINK is the correct mapping, according to your posts and what you told me about what was on various partitions. YOU SHOULD CHECK THESE BELOW TO MAKE SURE THEY ARE OK.

I only mapped the main OSes and stopped after the first project space, so you will need to fill in the remainder, but I think that should be straight forward for you to do.

Once you have checked and edited the content below, you need to replace the /media/pi/SETTINGS/installed_os.json file with that.  

Once you have done that, reboot PINN with your HDD connected and go to the maintenance menu.
Select (tick) the first Raspbian OS and click the FIX UP button. In the list, you can uncheck the first File System Check option, and check the Re-Run parition setup script option. Then execute it.

Now go to the main menu, click on Exit and select the first Raspbian OS to boot. Check it does actually boot off your Hard Disk and you have your previous files.

If that works, you need to fix up each of your other OSes in the same way and check they each boot. Check your HDD lights flash to indicate it is loading off your HDD rather than the SD card.

Hopefully that should get all your previous OSes working and accessible again.
There is some more tidying up you could do on the SD card, but that's for another time.

Will be out for most of the afternoon, but post any progress./questions as you go and I may be able to answer them sporadically.

new Installed_os.json (partial - to be edited & checked):
```
[ 
  { "bootable" : true, 
	"description" : "A port of Debian Stretch for the Raspberry Pi (full desktop version)", 
	"folder" : "/settings/os/Raspbian", 
	"group" : "", 
	"icon" : "/settings/os/Raspbian/icon.png", 
	"name" : "Raspbian", 
	"partitions" : [ 
		"/dev/mmcblk0p6", 
		"/dev/sda8" 
	], 
	"password" : "raspberry", 
	"release_date" : "2018-10-09", 
	"supported_models" : [ "Pi Model", "Pi 2", "Pi Zero", "Pi 3", "Pi Compute Module 3" ], 
	"url" : "http://www.raspbian.org/", 
	"username" : "pi" 
}, 
{ 	"bootable" : true, 
	"description" : "Retropie2 v4.4 gaming distro using Emulation Station for RPi2/3", 
	"folder" : "/settings/os/Retropie2", 
	"group" : "Games", 
	"icon" : "/settings/os/Retropie2/icon.png", 
	"name" : "Retropie2", 
	"partitions" : [ 
		"/dev/mmcblk0p8", 
		"/dev/sda6" 
	], 
	"password" : "raspberry", 
	"release_date" : "2018-04-14", 
	"supported_models" : [ "Pi 2", "Pi 3" ], 
	"url" : "https://retropie.org.uk/", 
	"username" : "pi" 
}, 
{ 	"bootable" : true, 
	"description" : "A port of Debian Stretch for the Raspberry Pi (full desktop version)", 
	"folder" : "/settings/os/Raspbian", 
	"group" : "", 
	"icon" : "/settings/os/Raspbian/icon.png", 
	"name" : "Raspbian", 
	"partitions" : [ 
		"/dev/mmcblk0p10", 
		"/dev/sda11" 
	], 
	"password" : "raspberry", 
[... 47 lines cut ...]
	"password" : "libreelec", 
	"release_date" : "2018-04-13", 
	"supported_models" : [ "Pi 2", "Pi 3" ], 
	"url" : "", 
	"username" : "root" 
}, 
{ 	"bootable" : false, 
	"description" : "A Project Space to partition the SD card equally", 
	"folder" : "/mnt/os/ProjectSpace", 
	"group" : "Utility", 
	"icon" : "/mnt/os/ProjectSpace/ProjectSpace_-_1.png", 
	"name" : "ProjectSpace - 1", 
	"partitions" : [ 
		"/dev/mmcblk0p18", 
		"/dev/sda12" 
	], 
	"password" : "?", 
	"release_date" : "2018-05-14", 
	"supported_models" : [ "Pi Zero", "Pi Model", "Pi Compute Module", "Pi 2", "Pi 3" ], 
	"url" : "https://", 
	"username" : "?" 
}, 
{ 	"bootable" : false, 
"description" : "A Project Space to partition the SD card equally", "folder" : "/mnt/os/ProjectSpace", "group" : "Utility", "icon" : "/mnt/os/ProjectSpace/ProjectSpace_-_2.png", "name" : "ProjectSpace - 2", "partitions" : [ "/dev/mmcblk0p20", "/dev/mmcblk0p21" ], "password" : "?", "release_date" : "2018-05-14", "supported_models" : [ "Pi Zero", "Pi Model", "Pi Compute Module", "Pi 2", "Pi 3" ], "url" : "https://", "username" : "?" }, { "bootable" : false, "description" : "A Project Space to partition the SD card equally", "folder" : "/mnt/os/ProjectSpace", "group" : "Utility", "icon" : "/mnt/os/ProjectSpace/ProjectSpace_-_3.png", "name" : "ProjectSpace - 3", "partitions" : [ "/dev/mmcblk0p22", "/dev/mmcblk0p23" ], "password" : "?", "release_date" : "2018-05-14", "supported_models" : [ "Pi Zero", "Pi Model", "Pi Compute Module", "Pi 2", "Pi 3" ], "url" : "https://", "username" : "?" }, { "bootable" : false, "description" : "A Project Space to partition the SD card equally", "folder" : "/mnt/os/ProjectSpace", "group" : "Utility", "icon" : "/mnt/os/ProjectSpace/ProjectSpace_-_4.png", "name" : "ProjectSpace - 4", "partitions" : [ "/dev/mmcblk0p24", "/dev/mmcblk0p25" ], "password" : "?", "release_date" : "2018-05-14", "supported_models" : [ "Pi Zero", "Pi Model", "Pi Compute Module", "Pi 2", "Pi 3" ], "url" : "https://", "username" : "?" }, { "bootable" : false, "description" : "A Project Space to partition the SD card equally", "folder" : "/mnt/os/ProjectSpace", "group" : "Utility", "icon" : "/mnt/os/ProjectSpace/ProjectSpace_-_5.png", "name" : [... cut ...]  : "https://", "username" : "?" }, { "bootable" : false, "description" : "A Project Space to partition the SD card equally", "folder" : "/mnt/os/ProjectSpace", "group" : "Utility", "icon" : "/mnt/os/ProjectSpace/ProjectSpace_-_8.png", "name" : "ProjectSpace - 8", "partitions" : [ "/dev/mmcblk0p32", "/dev/mmcblk0p33" ], "password" : "?", "release_date" : "2018-05-14", "supported_models" : [ "Pi Zero", "Pi Model", "Pi Compute Module", "Pi 2", "Pi 3" ], "url" : "https://", "username" : "?" } ]​
```

procount, 2018-11-04:
> Does it matter what order the project spaces are in? here is df -h:
> /dev/sda6         69G   18G   47G  28% /media/pi/root1 /dev/sda13        64G   52M   60G   1% /media/pi/prjroot01 /dev/sda7         70G  6.2G   60G  10% /media/pi/root_gen1 /dev/sda8         75G   15G   57G  21% /media/pi/root0 /dev/sda10        64G   84M   60G   1% /media/pi/Storage1 /dev/sda12        64G  4.1G   56G   7% /media/pi/prjroot7 /dev/sda11        65G   25G   37G  41% /media/pi/root-rbp2 /dev/sda14        64G   52M   60G   1% /media/pi/prjroot11 /dev/sda15        64G   52M   60G   1% /media/pi/prjroot21 /dev/sda16        64G   52M   60G   1% /media/pi/prjroot31 /dev/sda17        64G   52M   60G   1% /media/pi/prjroot41 /dev/sda18        64G   52M   60G   1% /media/pi/prjroot51 /dev/sda19        66G   52M   62G   1% /media/pi/prjroot61 /dev/sda9         67G  2.9G   61G   5% /media/pi/root_raspx1
>
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/procount/pinn/issues/195#issuecomment-435683434",
> "url": "https://github.com/procount/pinn/issues/195#issuecomment-435683434",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> },
> {
> "@type": "MessageCard",
> "@context": "http://schema.org/extensions",
> "hideOriginalBody": "false",
> "originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
> "title": "Re: [procount/pinn] How to restore dead SD card used to boot a HDD? (#195)",
> "sections": [
> [... 51 lines cut ...]
> }
> ],
> "themeColor": "26292E"
> }
> ]

No not really but it might make it more consistent to put them in the same order.

darkplatinum, 2018-11-04:
Ok, Ill continue placing them in some sort of order.

darkplatinum, 2018-11-04:
Im doing the final checks and i think i see issues with the mmcblk numbers. let me double check

darkplatinum, 2018-11-04:
Can you double check as well to make sure the code looks ok?

[ 
  { "bootable" : true, 
	"description" : "A port of Debian Stretch for the Raspberry Pi (full desktop version)", 
	"folder" : "/settings/os/Raspbian", 
	"group" : "", 
	"icon" : "/settings/os/Raspbian/icon.png", 
	"name" : "Raspbian", 
	"partitions" : [ 
		"/dev/mmcblk0p6", 
		"/dev/sda8" 
	], 
	"password" : "raspberry", 
	"release_date" : "2018-10-09", 
	"supported_models" : [ "Pi Model", "Pi 2", "Pi Zero", "Pi 3", "Pi Compute Module 3" ], 
	"url" : "http://www.raspbian.org/", 
	"username" : "pi" 
}, 
{ 	"bootable" : true, 
	"description" : "Retropie2 v4.4 gaming distro using Emulation Station for RPi2/3", 
	"folder" : "/settings/os/Retropie2", 
	"group" : "Games", 
	"icon" : "/settings/os/Retropie2/icon.png", 
	"name" : "Retropie2", 
	"partitions" : [ 
		"/dev/mmcblk0p8", 
		"/dev/sda6" 
	], 
	"password" : "raspberry", 
	"release_date" : "2018-04-14", 
	"supported_models" : [ "Pi 2", "Pi 3" ], 
	"url" : "https://retropie.org.uk/", 
	"username" : "pi" 
}, 
{ 	"bootable" : true, 
	"description" : "A port of Debian Stretch for the Raspberry Pi (full desktop version)", 
	"folder" : "/settings/os/Raspbian", 
	"group" : "", 
	"icon" : "/settings/os/Raspbian/icon.png", 
	"name" : "Raspbian", 
	"partitions" : [ 
		"/dev/mmcblk0p10", 
		"/dev/sda11" 
	], 
	"password" : "raspberry", 
	"release_date" : "2018-10-09", 
	"supported_models" : [ "Pi Model", "Pi 2", "Pi Zero", "Pi 3", "Pi Compute Module 3" ], 
	"url" : "http://www.raspbian.org/", 
	"username" : "pi" 
}, 
{ 	"bootable" : true, 
	"description" : "64-bit Gentoo Linux for the RPi3, with Xfce4 desktop", 
	"folder" : "/settings/os/gentoo64", 
	"group" : "General", 
	"icon" : "/settings/os/gentoo64/icon.png", 
	"name" : "gentoo64", 
	"partitions" : [ 
		"/dev/mmcblk0p12", 
		"/dev/sda7" 
	], 
	"password" : "raspberrypi64", 
	"release_date" : "2018-09-26", 
	"supported_models" : [ "Pi 3" ], 
	"url" : "https://github.com/sakaki-/gentoo-on-rpi3-64bit", 
	"username" : "demouser" 
}, 
{ 	"bootable" : true, 
	"description" : "Raspex with LXDE based on Ubuntu 18.04 LTS",
	"folder" : "/settings/os/Raspex", 
[... 134 lines cut ...]
		"/dev/sda18" 
	], 
	"password" : "?", 
	"release_date" : "2018-05-14", 
	"supported_models" : [ "Pi Zero", "Pi Model", "Pi Compute Module", "Pi 2", "Pi 3" ], 
	"url" : "https://", 
	"username" : "?" 
}, 
{ 	"bootable" : false, 
	"description" : "A Project Space to partition the SD card equally", 
	"folder" : "/mnt/os/ProjectSpace", 
	"group" : "Utility", 
	"icon" : "/mnt/os/ProjectSpace/ProjectSpace_-_8.png", 
	"name" : "ProjectSpace - 8", 
	"partitions" : [ 
		"/dev/mmcblk0p32", 
		"/dev/sda19" 
	], 
	"password" : "?", 
	"release_date" : "2018-05-14", 
	"supported_models" : [ "Pi Zero", "Pi Model", "Pi Compute Module", "Pi 2", "Pi 3" ], 
	"url" : "https://", 
	"username" : "?" 
} 
]​

darkplatinum, 2018-11-04:
No idea why te project space 1 is going into code format even though i did not select it.

darkplatinum, 2018-11-04:
Does it look like  }, and {  need to be the same indentation? My original os json file shows it?

procount, 2018-11-04:
> No idea why te project space 1 is going into code format even though i did not select it.
You need to put the whole file into code blocks to avoid this. Attaching the file by dragging it onto the post may be better. JSON parsers are a bit pinickity about the format (e.g. missing commas etc) but indentation should not matter.

The original file you posted did not retain any formatiing, so I adjusted it a bit to make it readable. 
If there is a problem with the formatting you could start again with your original file and just replace the 2nd partition file for each OS to map it to the HDD as I did.

I'll have a look at your last posted file now I'm home.

darkplatinum, 2018-11-04:
And how do I save the .json file? When i try to save it, it says "Can't open file to write". If you post my posted file after making any changes to it. Thanks.

procount, 2018-11-04:
It should not be necessary to login as root, but it might be mounted readonly.
In a terminal window try
```
sudo mount -o remount,rw  /media/pi/SETTINGS
```
and then try copying it again.
You can do it from the terminal with something like:
```
cp /home/pi/<desktop?>/installed_os.json  /media/pi/SETTINGS
```
No harm in adding a sudo at the front if that doesn't work.

darkplatinum, 2018-11-04:
So I can go ahead and reboot the Pi now?

procount, 2018-11-04:
Yep
Then you need to rerun the partition_setup script from the fix up ment on one Raspbian OS to test it works.

darkplatinum, 2018-11-04:
Ok here goes nothing... will report back...

procount, 2018-11-04:
I forgot to say - Before you re-run the partition setup script - set the destination drive to the HDD.

darkplatinum, 2018-11-04:
Raspbian 1 ok, raspbian 2 ok

darkplatinum, 2018-11-04:
Raspbian 1 ok, raspbian 2, I'm getting an error, "failed to start kernel modules" the mouse and keyboard not working.

darkplatinum, 2018-11-04:
The GUI loads up on raspbian 2 but I get that error on boot

procount, 2018-11-04:
But you don't get the error on Raspbian 1?

darkplatinum, 2018-11-04:
Ok, I ran the partition setup with the other OS. The retropie setup succeeded, Gentoo failed, LIBRE ELEC suceeded, raspex suceeded, all project spaces say cannot find setup script for projects space, Maybe insert source USB stick.

darkplatinum, 2018-11-04:
No error on raspbian 1 mouse and keyboard working fine on it.

darkplatinum, 2018-11-04:
Retropie also recieve ing kernel module error, keyboard not working.

procount, 2018-11-04:
There's no setup script for the projectspaces, cos they have no OS in them.
I wonder, did we get Raspbian2 and gentoo partitions the wrong way round?
Maybe worth checking them again.
How long ago did you install these OSes to your HDD? Are they still the same version as the versions you installed the boot partitions from?

darkplatinum, 2018-11-04:
TBH I don't really need Gentoo or LibreELEC as I don't have any data in them, the only ones I really need are the raspbians and retropie. My retropie was installed previously from an ad card using your dual boot tutorial.

darkplatinum, 2018-11-04:
Should I try fixing the retro and raspbian using filesystem check?

darkplatinum, 2018-11-04:
And btw I am not sure about their version.

darkplatinum, 2018-11-04:
RetroPie is probably not the latest version. Same as raspbian 2

procount, 2018-11-04:
Do you still have the original SD card that you installed Retropie from? If so, you could copy that Boot partition over the retropie boot partition you just installed to your new SD card. You perhaps need to do the same thing for Raspbian2.

procount, 2018-11-04:
Well, you have the general principle now of how to fix it, and the good news is you got your Raspbian1 working. Your issue now is just finding the correct version of boot partitions to match the original root partitions.

darkplatinum, 2018-11-04:
I do not have the SD card, I have probably wiped it and installed another OS for another Raspberry Pi.

procount, 2018-11-04:
Oh dear. Never mind. 
You can check the version of the kernel used in your Raspbian OSes (and probably others too) by looking on the root partition at /lib/modules and seeing what folders it has. Something like 4.14.71. check Retropie as well. The you can copy the matching kernel and boot image for it.

procount, 2018-11-05:
The latest Raspbian in NOOBS is 4.14.71, so I guess you may have started from an earlier version (4.14.50) and upgraded?
Likewise, the latest Retropie is 4.14.30, so the presence of 4.14.70 suggests you have upgraded that as well.

The main idea is to restore the boot partitions on your SD card to what existed on your SD card when it went bad. What we did up to now was to restore the boot partitions with the very latest version, which I assumed you had, but it seems to have caused some issues. So here are some proposals for you to get back to your original working OSes on your HDD.

Option1
---------

- Remove your HDD, and restore your installed_os.json file to what it was with your new installations, so they run off your SD card. (Keep the existing installed_os.json because you'll need it again later).
- Replace the OSes that didn't work with an older version.  You might have to construct your own os_list_v3.json file and put it on the PINN partition (/dev/mmcblk0p1) to reference such an old version. You can find the old Raspbian OSes at http://downloads.raspberrypi.org/raspbian/archive/.... Also you should disable all other download sources so you only get the specific versions you want, and they are not overridden by later versions.
- (You can use `uname -a` to see what version your kernel is. I think it would get this from the kernel itself rather than any info on the rootfs, but you'd need to check)
- Separately boot into the OSes that didn't work and upgrade them to the same version as is on your HDD. You may have to use rpi-update and pick a github revision to exactly match the kernel version on your HDD. Just doing `sudo dist-upgrade` may bring you to a version that is too new.
- Finally go back to your modified installed_os.json file to see if the upgraded boot partitions now match the rootfs kernel version and work correctly.

Option 2
----------
For Retropie, I can't imagine you have done much customisation, apart from adding some ROMS etc. So if the above doesn't work, or takes too long to do, you might be better to cut your losses by installing a new version of Retropie (maybe into one of your free ProjectSpaces) and just copying your ROMS and any other data files across. Perhaps you can do the same with Raspbian

darkplatinum, 2018-11-05:
And how do i get my original installed json file back? It was replace when i did the cp command earlier.

[33 more replies]

closed 2019-03-15 by procount