procount/pinn #196: Question - network automated install, howto ?

procount, 2018-11-08:
From fthiebolt (https://github.com/raspberrypi/noobs/issues/316)

@procount hello, i'm currently giving a try to PiNN to automatically install an OS through the network. I downloaded pinn-lite and modified recovery.cmdline like this:

runinstaller quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline alt_image_source=http://xxxxxxxxxxxx/images/os_list.json keyboard=fr silentinstall select="Raspbian,waitnetwork" no_default_source no_update nofirmware ssh lang=en

RPi boot, i can see the 'Raspbian' from our server in the PiNN menu but it won't automatically start its installation ... missing option ?? :) Thanks

procount, 2018-11-08:
Sorry, I found a small bug that was preventing it installing from installing.. I will release a new version shortly.

procount, 2018-11-09:
I think I have resolved the issue now. 
I have pushed a new version 2.8.5.6 to sourceforge testing for you to test and confirm it works.
Please download it from https://sourceforge.net/projects/pinn/files/testing/pinn-2856.zip/download and let me know if it now works, before I update the documentation and release it.

fthiebolt, 2018-11-12:
It works!!  ... distribution is automatically selected and install starts immediately ... finally, i spent so much time patching NOOBS to reach the same behaviour ... but PiNN is much more powerfull :)
One issue i have had with NOOBS is that sometime i was stuck during install ... is there a watchdog to restart the whole process ?
In your docs, you talk about a single file to remove to start a whole reinstall ... ?

For you to know the way we are using  our RPis: each RPi configuration is undertaken in a fully automated way ... if something goes wrong or just to upgrade to a newer release, we simply force a full reinstall (we don't care about existing partitions). To do this, we forcefully write the 64MB NOOBS partition on /dev/mmcblk0 (i.e disk) and restart RPi

Great job :)

fthiebolt, 2018-11-12:
One additional question:

it happens sometime that network is not available ... to let PiNN waiting for ever the network to arrive, can i specifiy something like 'networktimeout=0' or 'networktimeout=-1' ?

Thanks

procount, 2018-11-12:
> One issue i have had with NOOBS is that sometime i was stuck during install ... is there a watchdog to restart the whole process ?

I only ever experienced this on the 3B+ when it first came out, but I've not experienced it since. I assume the issue was resolved. 

The problem with a watchdog within PINN, is that the watchdog might be blocked itself. Better to run a watchdog outside of PINN if you need one.
You could put a script in background.sh or pinn_inst.sh and monitor /tmp/progress - it will increment in >=1% steps - then reboot (use `rebootp`) if it doesn't progress quickly enough. See https://github.com/procount/pinn/blob/master/README_PINN.md#installation-progress

> In your docs, you talk about a single file to remove to start a whole reinstall ... ?

I don't recall - can you provide a link?

> it happens sometime that network is not available ... to let PiNN waiting for ever the network to arrive, can i specifiy something like 'networktimeout=0' or 'networktimeout=-1' ?

You mean you want an extension of networktimeout so that it _will_ wait indefinitely?

procount, 2018-11-12:
> To do this, we forcefully write the 64MB NOOBS partition on /dev/mmcblk0 (i.e disk) and restart RPi

You can probably just get away with re-writing recovery.cmdline so that it includes `runinstaller`, but it depends on how confident you are on the rest of the installation being corrupted.

fthiebolt, 2018-11-12:
Hello,

 
> > One issue i have had with NOOBS is that sometime i was stuck during install ... is there a watchdog to restart the whole process ?
> 
> I only ever experienced this on the 3B+ when it first came out, but I've not experienced it since. I assume the issue was resolved. 

No, i mean that it is not uncommon to experience a network failure ... so having a kind of 'dirty_flag' means the system has not been completly installed last time ... so it will automatically restart the whole process

> The problem with a watchdog within PINN, is that the watchdog might be blocked itself. Better to run a watchdog outside of PINN if you need one.
> You could put a script in background.sh or [...]

ah, nice feature :)

> 
> > In your docs, you talk about a single file to remove to start a whole reinstall ... ?
> 
> I don't recall - can you provide a link?

"It is also possible for PINN to be bypassed altogether and one of the OSes booted directly from startup. In this way it is not even noticeable that PINN is there, but it can be reinstated simply by deleting a single file on the PINN partition."

> 
> > it happens sometime that network is not available ... to let PiNN waiting for ever the network to arrive, can i specifiy something like 'networktimeout=0' or 'networktimeout=-1' ?
> 
> You mean you want an extension of networktimeout so that it _will_ wait indefinitely?

exactly, for example when you connect a new RPi to the network and you had forgotten to set its IP in a DHCP server ... you're really happy it won't get stuck for ever (yes, we have RPi hidden in ceilling and others almost unreachable place)

Cheers
François

> 
> 
> https://github.com/procount/pinn/issues/196#issuecomment-437890244

procount, 2018-11-12:
> No, i mean that it is not uncommon to experience a network failure ... so having a kind of 'dirty_flag' means the system has not been completly installed last time ... so it will automatically restart the whole process

Have you tried this with PINN's silentinstall feature? I have a feeling that unless the install completes successfully, it won't be added to the list. So on a reboot, it may reinstall until it works. You may also find the silentreinstallnewer feature useful for your use cases as it may avoid you having to rewrite the 64MB PINN partition to reinstall - just make the new OS version available and reboot.

> "It is also possible for PINN to be bypassed altogether and one of the OSes booted directly from startup. In this way it is not even noticeable that PINN is there, but it can be reinstated simply by deleting a single file on the PINN partition."

This refers to the "autoboot.txt" file that specifies a partition to boot directly. This is done by the firmware, not PINN. So it prevents PINN from doing any of this silentinstall feature.
Deleting this file causes PINN to be executed as normal on boot again, instead of the OS.

fthiebolt, 2018-11-13:
Among things we discussed, the most important one is the ad-vitae eternam ability to wait for the network (i.e networktimeout=0 or whatever you want) because i ought not to get stuck with 'bricked' RPi waiting for ever that someone select a local OS install ;)

Merci

procount, 2018-11-13:
Use timeout=0
https://sourceforge.net/projects/pinn/files/testing/pinn-2856a.zip/download
(untested)

fthiebolt, 2018-11-13:
Hello,

you mean networktimeout=0 ?

F

procount, 2018-11-13:
Sorry, yes. networktimeout

fthiebolt, 2018-11-13:
it works too ... you saved me hours of hacks in nOObs, thanks :)

... and maybe one day, you'll also add an option 'watchdogInstall=yes' that will reboot the board in case install does not progress anymore ;)

et merci beaucoup de votre aide!
François

procount, 2018-11-13:
Je vous en prie!

procount, 2018-11-14:
This is now released in p2.8.5.7

closed 2018-11-14 by procount

eroom1966, 2019-11-03:
Can I ask for an example of how this is done ?
I have tried to reproduce, but it will not perform the silentinstall - I am clearly missing something 
... help !!

Here is a dropbox link to my modified version of pinn-lite
https://www.dropbox.com/sh/acmskpcarlcr73r/AADA7Q0EJJb_Il9pj6q8Quq1a?dl=0

This links to my os_list.json here
http://pilotaware.lode.co.uk/downloads/OGN/net/os_list.json

and my recovery.cmdline is

runinstaller quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline loglevel=2 sdhci.debug_quirks2=4 alt_image_source=http://pilotaware.lode.co.uk/downloads/OGN/net/os_list.json select="OGNR" silentinstall

But it does not autoinstall, it goes interactive - clearly doing something wrong, but not sure what

Thx
Lee

fthiebolt, 2019-11-08:
Hello,

On my side, it works very well :)

Here the applied recovery.cmdline:
'''
runinstaller quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline alt_image_source=http://<your server>/<some path>/os_list.json keyboard=fr silentinstall select="Raspbian,waitnetwork" networktimeout=0 no_default_source no_update nofirmware ssh lang=en loglevel=2 sdhci.debug_quirks2=4
'''
F