procount/pinn #197: PINN does not boot from usb

swekley, 2018-11-12:
PINN works when i copy the files on a fat32 formatted sd card, but I get the splash screen after 2 minutes when trying it with my SanDisk Extreme Pro usb stick.

I'm talking about this one: https://www.sandisk.com/home/usb-flash/extremepro-usb

Raspbian runs fine from it, however I don't want to stick with just raspbian on a 128gb stick and I'd prefer not to use any sd cards

I'm using a rpi 3b with the usb boot bit set

swekley, 2018-11-12:
Yes, well the screen stays black for 2 minutes and then it gives the splash screen. I'll try the Etcher image. USB is correctly formatted as fat32 using osx terminal's diskutil

swekley, 2018-11-12:
Using the Etcher image it's working fine now.
Thanks!

closed 2018-11-12 by swekley

lurch, 2018-11-12:
> which is a bit harder to achieve on a 128GB USB stick as the default for that is exFAT

See https://www.raspberrypi.org/documentation/installation/sdxc_formatting.md :slightly_smiling_face:

lurch, 2018-11-12:
ping @XECDesign and @ghollingworth
Does it seem plausible that a USB-boot would take two minutes just to get to the splash screen?

XECDesign, 2018-11-12:
I don't know if 2 minutes is right, but it would take a while. Remember the bootrom has fairly minimal code to enumerate USB devices, find and load start.elf block by block (which is relatively large). It's certainly not optimised for speed. Then add the time it takes for it to stop trying to boot from the SD card.

ghollingworth, 2018-11-12:
It shouldn't take that long, but it's possible it's doing something else (i.e. waiting for HDMI for example).  You could enable BOOT_UART and look at the UART output to see what is happening...

https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/

procount, 2018-11-12:
For comparison, I flashed PINN to my Sandisk Ultra USB 3.0 16GB USB Stick with Etcher
I booted it on my Pi 3B with no SD card fitted. 

10 secs to rainbow splash screen.
29 secs til the splash screen disappeared and the screen went black
34 secs for PINN's Raspberry logo to display & card formatting to start.
43 secs for PINN's recovery menu to come up and start scanning for OSes

I then re-formatted the whole stick as FAT 32 with win32diskimager and copied the PINN files across manually and timed the boot.

10 secs to rainbow splash screen.
26 secs til the splash screen disappeared and the screen went black
31 secs for PINN's Raspberry logo to display & card formatting to start.
39 secs for PINN's recovery menu to come up and start scanning for OSes

Timing was only by my watch, but nowhere near 2 minutes,