procount/pinn #203: Pi won’t boot upon an attempt to install

jtodom04, 2018-12-09:
When I try to boot the pi, nothing comes up just a black screen. Trying to do a fresh install.

procount, 2018-12-09:
You've not given much to go on, like what size sd card, how did you format it, which PINN version did you use, how did you write it, did any LEDs flash, etc.?

If you're struggling to do this, download the etcher image version and use Etcher to flash it to your sd card.

jtodom04, 2018-12-09:
I believe Etcher is for iso's, img's, etc? I am using a 25gb SD, using [PINN 2.8.5.7](
https://github.com/procount/pinn/releases/tag/p2.8.5.7). I guess if etcher was supposed to solve the problem it can't. It only uses archive images.

procount, 2018-12-10:
There is a special version of PINN that is an image file suitable for installation by Etcher.
You can download it from:
https://sourceforge.net/projects/pinn/files/Etcher/pinn-lite.img.zip/download

lurch, 2018-12-10:
AFAIK 25gb SD cards don't exist? They're always a power-of-2, i.e. 4gb, 8gb, 16gb, 32gb, 64gb, etc.

> You've not given much to go on, like what size sd card, how did you format it, which PINN version did you use, how did you write it, did any LEDs flash, etc.?

...and which model of Raspberry Pi you're using ;-)

procount, 2018-12-10:
This is probably a silly question, but the link you gave above to PINN 2.8.5.7 is to the SOURCE code of PINN. Did you actually download this and copy it to your SD card? That wouldn't actually work at all.

You should download the zip file from sourceforge: https://sourceforge.net/projects/pinn/files/pinn-lite.zip/download as detailed at the end of the 1st post in my Sticky forum thread: https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=142574
Or see the Quick-Start instructions here ->https://github.com/procount/pinn#quick-start
or here ->https://github.com/procount/pinn/blob/master/README_PINN.md#quick-start (This latter one includes how to install using Etcher using the PINN etcher image)

jtodom04, 2018-12-10:
Dang ok lemme delete my GitHub account I was downloading source code. Excuse my ignorance. I apologize.

procount, 2018-12-13:
No problem. I assume this issue is (or will be) resolved by downloadng the correct zip file from sourceforge. So I'll close this issue. Please feel free to reopen it if you still have problems.

closed 2018-12-13 by procount