procount/pinn #204: How to set Background.sh.??

biswaKL, 2018-12-11:
Can you please elaborate where to add background.sh ?? i tried to add it to pinn via ssh but it says read-only file system. Where can i define that??
I want to blink a LED until the progress reaches 100%.

procount, 2018-12-11:
PINN mounts the recovery partition (/dev/mmcblk0p1) on /mnt as a read-only file system, because it is not normally intended to modify anything on this partition. After all, it was the original home of the installation files for Raspbian to be used in the event of a recovery situation. It was one of NOOBS' golden rules - never to modify this partition, but this has since been relaxed somewhat, especially by PINN.

So, if you want to modify anything on this partition, you need to make it writeable first.
```
mount -o remount,rw /mnt
cd /mnt
```
Now you should be able to add your background.sh file to monitor the /tmp/progress file. 
When you've done that, don't forget to `sync` before rebooting, or you will lose your new files.
If you want to make the partition read-only again you can do:
```
mount -o remount,rw /mnt
```
Also note that the progress percentage in /tmp/progress is never precise, because it is monitoring all I/O requests on the target drive. So sometimes when it reaches 100%, there is still some writing to be done before it is finished. On the other hand, some OSes may have finished installing before the progress reaches 100%, so 100% is never reached. In a future release I may consider forcing a 100% output in this case for scripts such as yours, so that they can actually detect the end of the installation.

biswaKL, 2018-12-11:
so basically, if i put my background.sh in RECOVERY then it will load automatically is it?

procount, 2018-12-11:
Yes.
Of course, if you have a Windows PC and plug your SD card into it via a USB card reader or something, then you can just drop your background.sh script onto the RECOVERY partition, just like when you first install PINN from a zip file.

biswaKL, 2018-12-11:
Cool great, working like a charm, 100% issue is not a major stuff, i can still use it. Tanks for the quick update.

Fateofiko, 2019-03-22:
Hi!
I am trying to do the same thing, but my script do not start. The script works if i run it manually from the raspberry OS, but do not start automatically, when i copy it to the PINNs directory!
Is there some options in the recovery.cmdline file, that can block the background.sh to be started?
My recovery file contains:
runinstaller quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline repo_list=http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json no_update reserve=+2000 disableusbimages no_default_source silentinstall

procount, 2019-03-22:
No there shouldn't be anything that blocks it. You need to be aware of some things though:
1. It uses the busybox shell (#!/bin/sh) rather than bash
2. the recovery partition it runs from is mounted at /tmp/mnt (IIRC)
3. If you want to run it continuously in the background, it's probably best to copy itself (or another script) to the /tmp folder and run it from there. especially if forking to the background, to avoid preventing the recovery partition from being unmounted/mounted.

There is also pinn_init.sh that can be used. It is executed before PINN is started, but the same caveats apply.

Fateofiko, 2019-03-25:
Hi @procount ,
I changed my script to use the busybox shell (#!/bin/sh), and now it runs, but as you said it causes the recovery partition to be unmounted/mounted.
So i writen the background script to copy another script to /tmp and execute it, i am not sure why, but it does not work!
Is there more detailed documentation about this background.sh script, because i wasn't able to find such? And, i do not want to bother you with stupid questions :) !

procount, 2019-03-25:
Try posting your 2 scripts here and I'll take a look. Tbh I 'm getting a bit rusty on this feature so it would be a good refresher for me.

Fateofiko, 2019-03-25:
Ok, this are my scripts:
Script one with name background.sh:
`
#!/bin/sh

TMP_DIR=/tmp/
MOUNT_DIR=/tmp/mnt/
SCRIPT=background_tmp.sh

cp -a "$MOUNT_DIR$SCRIPT" "$TMP_DIR"
. "$TMP_DIR$SCRIPT"
`
In the first script i also tried and this :        . "$TMP_DIR$SCRIPT" &
to start the script in background :)!

And the second script with name background_tmp.sh:
`
#!/bin/sh

BASE_GPIO_PATH=/sys/class/gpio

# Assign names to GPIO pin numbers for each light
RED=9

# Assign names to states
ON="1"
OFF="0"

  if [ ! -e $BASE_GPIO_PATH/gpio$RED ]; then
    echo "$RED" > $BASE_GPIO_PATH/export
  fi

  echo "out" > $BASE_GPIO_PATH/gpio$RED/direction

# Loop forever until user presses Ctrl-C
while [ 1 ]
do
  # Red on
  echo $ON > $BASE_GPIO_PATH/gpio$RED/value
  sleep 1

  # Red off
  echo $OFF > $BASE_GPIO_PATH/gpio$RED/value
  sleep 1
done

`

This two scripts are just test scripts, the idea is just to make a diode to blink while the progress bar of the installation process is moving. When this starts to work, i will make the diode to blink every time the percentage increase with 1%!

procount, 2019-03-25:
1. I fixed the formatting of your code 😄 
2. I was wrong 😢  background.sh is run from /mnt (I updated my previous comment accordingly)

lurch, 2019-03-25:
@Fateofiko If you can, it might be worth enabling SHH in PINN and connecting to PINN over SSH (ethernet will be much easier than WiFi!) and manually running your `background_tmp.sh` to check that it actually works in the buildroot / busybox environment of PINN :slightly_smiling_face:

procount, 2019-03-25:
This seems to work:

background.sh
```
#!/bin/sh

TMP_DIR=/tmp/
MOUNT_DIR=/mnt/
SCRIPT=background_tmp.sh

cp -a "$MOUNT_DIR$SCRIPT" "$TMP_DIR"
"$TMP_DIR$SCRIPT" &
```

background_tmp.sh
```
#!/bin/sh

BASE_GPIO_PATH=/sys/class/gpio

# Assign names to GPIO pin numbers for each light
RED=9

# Assign names to states
ON="1"
OFF="0"

  if [ ! -e $BASE_GPIO_PATH/gpio$RED ]; then
    echo "$RED" > $BASE_GPIO_PATH/export
  fi

  echo "out" > $BASE_GPIO_PATH/gpio$RED/direction

# Loop forever until user presses Ctrl-C
while [ 1 ]
do
  # Red on
  echo $ON > $BASE_GPIO_PATH/gpio$RED/value
  sleep 1

  # Red off
  echo $OFF > $BASE_GPIO_PATH/gpio$RED/value
  sleep 1
done
```

closed 2019-03-25 by procount

Fateofiko, 2019-03-26:
Hi @lurch, thanks for the good advice, but i use ssh only for testing, everything should work automatically :) !
@procount , thanks for the help, i changed the MOUNT_DIR to /mnt/ and now everything works perfect.
One feature in the first script, you are using this line " "$TMP_DIR$SCRIPT" & " , but i am using this line " . "$TMP_DIR$SCRIPT" & ", i prefer to include the dot ( . ), i am not linux guru, i have not test it without dot, but may be, it works equally and both ways :) ! 
I hope this scripts to be a good example for everyone, who wants to test this feature :) !

procount, 2019-03-26:
I think @Lurch was only suggesting ssh for testing purposes to find out what was not working. It's exactly what I did.

AIUI, '.' sources the 2nd file (like include) so it runs in the same shell, whereas omitting it starts a new shell, In either case, the '&' is necessary to put it in the background.

lurch, 2019-03-26:
> but i use ssh only for testing

Yes, that's exactly what I was suggesting: using it for testing that your `background_tmp.sh` works correctly with the busybox shell, and that it doesn't use any bash-specific features that aren't in PINN :wink: 
Once you've verified that `background_tmp.sh` itself works, you can then focus on integrating it with PINN's `background.sh` feature.

> i prefer to include the dot ( . ), i am not linux guru

The dot at the front means "source this script", whereas without the dot it means "execute this script". More info [here](https://superuser.com/questions/176783/what-is-the-difference-between-executing-a-bash-script-vs-sourcing-it)

lurch, 2019-03-26:
> I hope this scripts to be a good example for everyone, who wants to test this feature :) !

It's just a style thing (and doesn't really matter), but I personally tend to find e.g.
```bash
TMP_DIR=/tmp
MOUNT_DIR=/mnt
SCRIPT=background_tmp.sh
cp -a "$MOUNT_DIR/$SCRIPT" "$TMP_DIR/"
```
a little bit more readable than 
```bash
TMP_DIR=/tmp/
MOUNT_DIR=/mnt/
SCRIPT=background_tmp.sh
cp -a "$MOUNT_DIR$SCRIPT" "$TMP_DIR"
```
(and it also makes it easier to avoid accidentally putting two consecutive slashes into a path)
:man_shrugging:

Fateofiko, 2019-03-26:
Hi,
I am not sure that i have to open a new issue, but i will ask here (because i will use this in the background.sh script ), is there any other tool in PINN except "telnet" , which i can use for sending http request to some host?

procount, 2019-03-26:
Wget.
Not sure about curl - can't remember but you csn try it.

Fateofiko, 2019-03-26:
I already tried curl but the output is "command not found"!
Yes, wget is a good idea!
Thanks :) !!!

giuspe01, 2019-09-16:
hi, i'm trying to make sure that when you enter recovery mode (via gpio) you start the raspbian system, i tried to use both background.sh and pinn_init.sh but none of the 2 works, i also noticed that pinn_init.sh it always runs even without entering recovery so I'm avoiding this method and I'm trying to use background.sh but it doesn't work my file contains only "rebootp 8", I also tried to run a second file but it doesn't work the same. Can anyone help me?

giuspe01, 2019-09-16:
I opened a issues #317