procount/pinn #205: Error in Creating Partition Table

biswaKL, 2018-12-11:
If i am running background.sh, and starting the installation i am getting **Error creating partition table**.

If i remove the background.sh, then it runs fine.
Here is what is have in background.sh

`#!/bin/sh
file=/tmp/progress

LEDPIN=9
OFF=0
ON=1

echo $LEDPIN > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio$LEDPIN/direction
#trap 'cleanup $LEDPIN' 1 2 15
echo $ON > /sys/class/gpio/gpio$LEDPIN/value    # turn on

while true
do
    if [ -f $file ]
        then
            i=0
            echo "File  found."
            while [ $i -lt 100 ]
            do
    
                echo $OFF > /sys/class/gpio/gpio$LEDPIN/value   # turn off
                sleep 1
    
                data="$(cat $file)"
                data=$(echo $data | sed 's/%/\n/g')
                for i in $data; do
                    if [ $i -le 100 ]; then
                        echo $i
                    fi
                done
    
                echo $ON > /sys/class/gpio/gpio$LEDPIN/value    # turn on
                sleep 1
    
            done
    fi
done
echo $OFF > /sys/class/gpio/gpio$LEDPIN/value    # turn on
`

My Code is a infinite loop which waits for the progress file to be created and after getting created it keeps checking the progress file value.

biswaKL, 2018-12-11:
**Re-reading the partition table ...
"
mount: /dev/mmcblk0p1 is already mounted or /mnt busy
       /dev/mmcblk0p1 is already mounted on /mnt
Error: "Error creating partition table
"**

This what is got from debug log.

procount, 2018-12-11:
I was about to suggest moving it to /tmp, but I see you figured it out yourself.

I seem to recall there was a change in how NOOBs wrote the partition table that affected the operation of this script. I must update the documentation accordingly.