procount/pinn #206: Windows IoT Installation

OxygenGS, 2018-12-13:
Hey guys,

Well i installed the Insider versio for the PI 3B+.
Installing went ok, but when i start the IoT version on my PI, i get to the main menu so to say. WHere i can also go into the settings. However, problem is now: i cannot find any WiFi acces points in it, just says cannot find. How to fix this? i wanna use it wireless and NOT wired.

procount, 2018-12-13:
I'm afraid you'll have to ask Microsoft about that as they provide the images. NOOBS and PINN just install them. My guess is that MS will need to add the appropriate wifi driver, which I think changed for the 3B+ model.

matthuisman, 2018-12-14:
https://docs.microsoft.com/en-us/windows/iot-core/release-notes/insider/rpi3bp

> Known issues in this build:
> This image is only meant for RPi3B+ and will not boot on RPi2.
> F5 driver deployment from Visual Studio does not work on IoT Core.
> **Onboard WIFI and Bluetooth do not work on RPI3B+.**
> Ft5406 touch screen driver is disabled on RPi3B+.
> SD card activity LED is disabled.

closed 2019-03-15 by procount